meshtastic/firmware #2288: [Feature Request]: Custom duty cycle

rosmo, 2023-02-18:
### Platform

NRF52

### Description

If I'm understanding correctly, `lora.override_duty_cycle` disables duty cycle entirely, otherwise it is taken from the region settings. Would it be possible to add an option to set a lower duty cycle (eg. 1%)?

garthvh, 2023-02-18:
What would the use case be? The duty cycle is set by local regulations and can be overridden if you are a licensed ham.

caveman99, 2023-02-18:
At 1% Meshtastic is unusable.

closed 2023-02-19 by garthvh

wucke13, 2025-03-18:
>  The duty cycle is set by local regulations and can be overridden if you are a licensed ham.

Well there is more than one value as per the regulations. For example, [page 3, row 2](https://www.bundesnetzagentur.de/SharedDocs/Downloads/DE/Sachgebiete/Telekommunikation/Unternehmen_Institutionen/Frequenzen/Allgemeinzuteilungen/FunkanlagenGeringerReichweite/2018_05_SRD_pdf.pdf?__blob=publicationFile&v=1) allows for specific frequencies to be used with 500 mW but only up to 2.5% duty cycle.

TL;DR there might be more granularity to the regulation for non-ham folks than one duty-cycle per region. Hence an "advanced" option allowing to customize the duty-cycle would be appreciated.

fifieldt, 2025-03-18:
@wucke13 , do you happen to have a specific use-case, or is it about trying to make the mesh better overall?