meshtastic/firmware #2291: [Bug]: Router node not forwarding messages

Josh-SMU, 2023-02-20:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.17

### Description

When using two RAK4631 devices to send direct messages, we tested how far we could go from each other before being unable to send/receive messages. We then put another device in router-client mode closer than the last spot we had good signal to each other, and the other person went further out. This test was to see if the router node would increase our range. After various tests it turns out the router node isnt forwarding our messages at all. This seems to be related to #1459. We were using a custom channel/PSK and all three devices were on the same channel settings.

garthvh, 2023-02-20:
1459 has been fixed since before 2.0

Josh-SMU, 2023-02-20:
Yes however I am having the same issue, at least in practice. How can I see if the router node is forwarding the messages or not? Im not too familiar with using the CLI for message monitoring

Josh-SMU, 2023-02-20:
> > ### Relevant log output
> > _No response_
> 
> If you're going to claim it's not doing something, it would be helpful to include the necessary logs that corroborate that claim.

Yes, Im asking how to obtain those logs as my tests were using the Android app. I could very clearly pinpoint that the router node was "not doing anything" because if it were, the messages would have come in for either of us, even if the ACK wasnt being forwarded (which would just mean no read receipt)

GUVWAF, 2023-02-23:
This is working for me. 
First thing would be to check whether you have a direct connection to the router from either device. If it is "not doing anything", it might be that e.g. your channel settings were not saved properly.
To obtain the logs, easiest would be to [install the Python CLI](https://meshtastic.org/docs/software/python/cli/installation), connect your device and use `meshtastic --noproto`. You can also double check whether you configured it correctly by running `meshtastic --info`.

Josh-SMU, 2023-02-23:
Thank you. I saw that the channel setting set via the Android app didnt actually take so set it via CLI and I was then able to see messages flowing; however the previous channel was on the same mode (SHORT_FAST) and was not forwarding anything. There was nothing printed to the logs using --noproto with that channel setting as well. My understanding was that as long as a node is configured with the same modem settings (region and mode) it would forward packets sent on the same region and mode?

GUVWAF, 2023-02-23:
You saw nothing when using --noproto? It should at least print some logs related to battery power every once in a while, otherwise something else is wrong.
Yes, it should forward packets if the region and mode is the same.

Josh-SMU, 2023-02-23:
Sorry by nothing I meant only battery info. If its primary channel is using a different PSK should I still see that it received and forwarded a packet despite not being able to decrypt it? Because there was no output other than the battery stuff.

GUVWAF, 2023-02-23:
Yes, even with different PSK it should forward the packet.

Josh-SMU, 2023-02-23:
I have reset the node to a new channel with a new PSK on SHORT_FAST and connected with --noproto and again see the same behavior; nothing in the output aside from some battery stuff mostly; no reaction to sending messages to the channel on the other device.

Josh-SMU, 2023-02-23:
Actually now this is all Im seeing:

```
WARNING file:mesh_interface.py _sendToRadio line:511 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:389 Not sending packet because protocol use is disabled by noProto
```

GUVWAF, 2023-02-23:
Can you share the output of `meshtastic --info`?

Josh-SMU, 2023-02-23:
```
Connected to radio

Owner: Node1 (N1)
My info: { "myNodeNum": 4, "maxChannels": 8, "firmwareVersion": "2.0.17.5d1c06b", "errorCount": 3, "bitrate": 784.76825, "messageTimeoutMsec": 300000, "minAppVersion": 20300 }
Nodes in mesh:  {'num': 4, 'user': {'id': '!26484182', 'longName': 'Node1', 'shortName': 'N1', 'macaddr': 'f7:4d:26:48:41:82', 'hwModel': 'RAK4631'}, 'position': {'seqNumber': 1}, 'deviceMetrics': {'batteryLevel': 64, 'voltage': 3.872}}  {'num': 2731118512, 'user': {'id': '!a2c98fb0', 'longName': 'Node2', 'shortName': 'N2', 'macaddr': 'd2:c1:a2:c9:8f:b0', 'hwModel': 'RAK4631'}, 'position': {'latitudeI': redacted, 'longitudeI': redacted, 'time': 1672857371, 'latitude': redacted, 'longitude': redacted}, 'snr': 5.5, 'deviceMetrics': {'batteryLevel': 38, 'voltage': 3.63}}  {'num': 4235825937, 'user': {'id': '!fc799311', 'longName': 'Pocket1', 'shortName': 'Pkt1', 'macaddr': 'd7:7c:fc:79:93:11', 'hwModel': 'RAK4631'}, 'snr': 12.5}

Preferences: { "device": { "role": "ROUTER_CLIENT", "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 16 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "modemPreset": "SHORT_FAST", "txPower": 30 }, "bluetooth": { "enabled": true, "mode": "FIXED_PIN", "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=secret { "psk": "Hp+XC4tnxiyX6modified+0Or3PaDkqKZw5nK8=", "name": "TestChannel" }

Primary channel URL: https://meshtastic.org/e/#Ci8SIB6flwuLZ8Ysl-uGocmodifiedimcOZyvGgtUZXN0Q2hhbm5lbBIGCAEQBlAe
```

Josh-SMU, 2023-02-23:
After a reboot the battery stuff came back again, not sure why it's hiccuping

GUVWAF, 2023-02-23:
It has discovered the other nodes (Node2 and Pocket1), so it should have received their packets at some time. 
What seems odd to me is that it "myNodeNum" is 4, while normally this is taken from the MAC address. It seems that it tried to pick a [reserved NodeNum](https://github.com/meshtastic/firmware/blob/master/src/mesh/NodeDB.cpp#L371).

If you send from this node, do the others receive it?

Josh-SMU, 2023-02-23:
I didnt clear the NodeDB and those are from previous tests on the same channel

Josh-SMU, 2023-02-23:
I sent a test message from this node via cli on the primary channel (different from the other device but same mode) and got nothing.

Josh-SMU, 2023-02-23:
This is expected though due to them being on different channels correct? Did you mean send to a specific device? Or put it back on the same channel as the other device and try?

GUVWAF, 2023-02-23:
Where did you check if you got anything? In the logs it should show it received something it couldn't decrypt, but in the Android app it will not show anything in that case.
On the same channel it should also show it in the Android app.

Josh-SMU, 2023-02-23:
Sending from CLI on same primary channel as other device: Msg received on Android
Sending from Android on same primary channel as other device: Msg received via CLI
Sending from Android on different channel than other device: No response

GUVWAF, 2023-02-23:
So the original issue of "not doing anything" is solved at least.
I really can't think of anything other than not having the same LoRa settings if you don't get a response. For the last case you were looking in the logs I assume, when you did see other logging?

Josh-SMU, 2023-02-23:
For the last case it was just periodically dumping battery info, not sure entirely what you mean. There was no indication that it was printing anything at all due to a message being sent or at the same time. As far as LoRa settings I can confirm both devices are set to SHORT_FAST.

GUVWAF, 2023-02-23:
Yes, just to be sure it wasn't stuck like you had experienced before. 

At this point I don't know what is going wrong when you send on a different channel, I'm sorry. Since on the same channel you can now successfully send and receive directly between nodes, the router will also forward this packet if it was in between two nodes that are outside of range of each other.

andrekir, 2023-02-23:
> ```
> "lora": { "usePreset": true, "modemPreset": "SHORT_FAST", "txPower": 30 }
> ```

matching just modem preset is not enough. from above, you're missing the `region`. also, if `channel_num` is not set, primary channel names must match.

you can check all the radio configs in use from the logs when devices boot up.

Josh-SMU, 2023-02-23:
Can you help me understand? Originally I set a new channel from the Android app. Then I changed the channel on the problem device (router) from the Android app to a new name (so it would get a new PSK, etc). In the app, all of these devices show region as US. 

Does the Android app fail to set these settings then? Even when testing via CLI above, I changed channels by using --seturl from the url exported from the app.

How should I configure these three devices the correct way so that:

Device1 and Device2 are on the same channel with same PSK
Device3 is on a different channel with different PSK but can forward packets between Device1 and Device2

andrekir, 2023-02-24:
rebroadcast can't work with the settings you provided, it can only work when nodes lora config match.

Josh-SMU, 2023-02-24:
Ok after reading through the relevant wiki pages I have set both devices (Device1 and Node1 [Node1 should rebroadcast]) to the following:
```
"lora": { "usePreset": true, "modemPreset": "MEDIUM_FAST", "region": "US", "hopLimit": 3, "txEnabled": true, "txPower": 30, "channelNum": 1 },
```

But still when sending to the channel on Device1 and listening on Node1 (which is on a different channel) I see no response at all. And to clarify, by channel I mean "channel URL" which is the name and PSK; both devices are set to lora.channel_num 1

garthvh, 2023-02-24:
I would put the same channels on all three devices and then use the tracereoute from the python cli or iOS app to test where the packets are going.

closed 2023-02-24 by garthvh

Josh-SMU, 2023-02-24:
Why is this closed? I am not seeing the expected behavior so this appears to be a bug.

Josh-SMU, 2023-02-24:
Device1:

```
INFO  | ??:??:?? 2 Starting meshradio init...
INFO  | ??:??:?? 2 Radio freq=902.125, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=US, name=TESTMESH, config=4, ch=0, power=30
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 0
INFO  | ??:??:?? 2 Radio frequency: 902.125000
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=22
INFO  | ??:??:?? 2 SX126x init result 0
INFO  | ??:??:?? 2 Frequency set to 902.125000
INFO  | ??:??:?? 2 Bandwidth set to 250.000000
INFO  | ??:??:?? 2 Power output set to 22
DEBUG | ??:??:?? 2 Current limit set to 140.000000
DEBUG | ??:??:?? 2 Current limit set result 0
```

Device2 (the one thats not rebroadcasting):
```
INFO  | ??:??:?? 1 Starting meshradio init...
INFO  | ??:??:?? 1 Radio freq=902.125, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=US, name=TACMESH, config=4, ch=0, power=30
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 0
INFO  | ??:??:?? 1 Radio frequency: 902.125000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=22
INFO  | ??:??:?? 1 SX126x init result 0
INFO  | ??:??:?? 1 Frequency set to 902.125000
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 22
DEBUG | ??:??:?? 1 Current limit set to 140.000000
DEBUG | ??:??:?? 1 Current limit set result 0
```

Josh-SMU, 2023-02-24:
Device1 logs when sending message:
```
DEBUG | ??:??:?? 42 PACKET FROM PHONE (id=0xa631adc2 fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 42 handleReceived(USER) (id=0xa631adc2 fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 42 Module 'text' wantsPacket=1
INFO  | ??:??:?? 42 Received text msg from=0x0, id=0xa631adc2, msg=Test message
INFO  | ??:??:?? 42 Enter state: ON
DEBUG | ??:??:?? 42 Module 'text' considered
DEBUG | ??:??:?? 42 Module 'canned' wantsPacket=1
DEBUG | ??:??:?? 42 Module 'canned' considered
DEBUG | ??:??:?? 42 Module 'ExternalNotificationModule' wantsPacket=1
INFO  | ??:??:?? 42 External Notification Module Disabled
DEBUG | ??:??:?? 42 Module 'ExternalNotificationModule' considered
DEBUG | ??:??:?? 42 Module 'routing' wantsPacket=1
INFO  | ??:??:?? 42 Received routing from=0x0, id=0xa631adc2, portnum=1, payloadlen=12
DEBUG | ??:??:?? 42 Routing sniffing (id=0xa631adc2 fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 42 Module 'routing' considered
DEBUG | ??:??:?? 42 (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=32, time 211 ms
DEBUG | ??:??:?? 42 Setting next retransmission in 6434 msecs:  (id=0xa631adc2 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 42 Add packet record (id=0xa631adc2 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 42 Original length - 12 
DEBUG | ??:??:?? 42 Compressed length - 8 
DEBUG | ??:??:?? 42 Original message - Test message 
DEBUG | ??:??:?? 42 Using compressed message.
DEBUG | ??:??:?? 42 Using AES256 key!
DEBUG | ??:??:?? 42 Software encrypt fr=fc799311, num=a631adc2, numBytes=16!
DEBUG | ??:??:?? 42 enqueuing for send (id=0xa631adc2 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted)
DEBUG | ??:??:?? 42 txGood=2,rxGood=0,rxBad=0
INFO  | ??:??:?? 42 Telling client we have new packets 5
INFO  | ??:??:?? 42 BLE notify fromNum
DEBUG | ??:??:?? 42 [RadioIf] Starting low level send (id=0xa631adc2 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted priority=64)
DEBUG | ??:??:?? 42 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=32, time 211 ms
DEBUG | ??:??:?? 42 [RadioIf] AirTime - Packet transmitted : 211ms
INFO  | ??:??:?? 42 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 42 encoding toPhone packet to phone variant=11, 12 bytes
DEBUG | ??:??:?? 42 [RadioIf] Completed sending (id=0xa631adc2 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted priority=64)
```

GUVWAF, 2023-02-24:
That looks normal as well. 
I'm sorry, at this point I don't understand why this packet would not show in the logs of Device 2. For me, if it is sent on a different channel, I see this for the receiving node:
`No suitable channel found for decoding, hash was 0xf2!`

Josh-SMU, 2023-02-24:
@garthvh any idea? If this is not working for one person despite reflashing firmware then it should be considered a valid bug even if a few others are not seeing it.

Josh-SMU, 2023-02-24:
Ok what the heck, now suddenly its working:
```
DEBUG | ??:??:?? 81 [RadioIf] Lora RX (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
DEBUG | ??:??:?? 81 [RadioIf] AirTime - Packet received : 182ms
DEBUG | ??:??:?? 81 [Router] Add packet record (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
WARN  | ??:??:?? 81 [Router] No suitable channel found for decoding, hash was 0x2e!
DEBUG | ??:??:?? 81 [Router] packet decoding failed or skipped (no PSK?) (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
DEBUG | ??:??:?? 81 [Router] Module 'routing' wantsPacket=1
INFO  | ??:??:?? 81 [Router] Received routing from=0xfc799311, id=0xa631adc3, portnum=8, payloadlen=16129
DEBUG | ??:??:?? 81 [Router] Routing sniffing (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
INFO  | ??:??:?? 81 [Router] Rebroadcasting received floodmsg to neighbors
DEBUG | ??:??:?? 81 [Router] enqueuing for send (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=2 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
DEBUG | ??:??:?? 81 [Router] txGood=5,rxGood=6,rxBad=0
DEBUG | ??:??:?? 81 [Router] rx_snr found. hop_limit:2 rx_snr:10.750000
DEBUG | ??:??:?? 81 [Router] rx_snr found in packet. As a router, setting tx delay:420
DEBUG | ??:??:?? 81 [Router] Delivering rx packet (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
DEBUG | ??:??:?? 81 [Router] Forwarding to phone (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=3 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23)
WARN  | ??:??:?? 81 [Router] No suitable channel found for decoding, hash was 0x2e!
DEBUG | ??:??:?? 81 [Router] Module 'routing' considered
DEBUG | ??:??:?? 82 [RadioIf] Starting low level send (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=2 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23 priority=64)
DEBUG | ??:??:?? 82 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=24, time 182 ms
DEBUG | ??:??:?? 82 [RadioIf] AirTime - Packet transmitted : 182ms
DEBUG | ??:??:?? 82 [RadioIf] Completed sending (id=0xa631adc3 fr=0x11 to=0xff, WantAck=0, HopLim=2 Ch=0x2e encrypted rxSNR=10.75 rxRSSI=-23 priority=64)
```

Josh-SMU, 2023-02-24:
Ill open a new ticket concerning the stability as it works but only intermittently on boot sessions.