meshtastic/firmware #2296: [Bug]: RAK11300 radio not detected

D4rk4, 2023-02-21:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.0.20

### Description

The device has not booted up.

### Relevant log output

```Shell
INFO  | ??:??:?? 3 Power output set to 22
SPI::beginTransaction(clk=4000000, bo=MSB
                                         SPI: Reusing existing initted SPI
                                                                          SPI::transfer(0d), cpol=0, cpha=0
                                                                                                           SPI: read back 00
                                                                                                                            SPI::transfer(08), cpol=0, cpha=0
                                                                                                                                                             SPI: read back 00
 SPI::transfer(e7), cpol=0, cpha=0
                                  SPI: read back 00
                                                   SPI::transfer(38), cpol=0, cpha=0
                                                                                    SPI: read back 00
                                                                                                     SPI::endTransaction()
                                                                                                                          DEBUG | ??:??:?? 3 Current limit set to 140.000000
DEBUG | ??:??:?? 3 Current limit set result -2
SPI::beginTransaction(clk=4000000, bo=MSB
                                         SPI: Reusing existing initted SPI
                                                                          SPI::transfer(1d), cpol=0, cpha=0
                                                                                                           SPI: read back 00
                                                                                                                            SPI::transfer(08), cpol=0, cpha=0
                                                                                                                                                             SPI: read back 00
 SPI::transfer(ac), cpol=0, cpha=0
                                  SPI: read back 00
                                                   SPI::transfer(00), cpol=0, cpha=0
                                                                                    SPI: read back 00
                                                                                                     SPI::endTransaction()
                                                                                                                          INFO  | ??:??:?? 3 Set Rx Power Saving Gain mode; result: 65534
WARN  | ??:??:?? 3 Failed to find SX1262 radio
ERROR | ??:??:?? 3 NOTE! Recording critical error 3 at src/main.cpp:513
```

thebentern, 2023-02-22:
Is this new or does it occur on previous releases. Also which base board?

D4rk4, 2023-02-22:
As I know RAK11300 is never working, the baseboard is RAK4631-mPCIe.R.1.0 (dummy core board with USB over mPCIe). I guess we need a dedicated variant for RAK11300.

P.S. RAK11300 not booting at all on RAK19003 base board =)

caveman99, 2023-02-22:
RAK11300 is the RP2040+SX1262 core. The RP2040 is actually starting up?

D4rk4, 2023-02-22:
@caveman99 yes it's starting, but halted on SX1262 initialization. Look like an SPI issue or wrong pins?

caveman99, 2023-02-22:
Probably. the pico definition is for a Raspberry Pico with Waveshare SX1276 shield. This needs a new target for sure.

caveman99, 2023-04-02:
please reopen if this is still an issue with current master - we pulled in an Radiolib init fix today.

thebentern, 2023-04-02:
Still not initializing, sadly
`INFO  | ??:??:?? 10 SX126x init result -2
WARN  | ??:??:?? 10 Failed to find SX1262 radio
ERROR | ??:??:?? 10 NOTE! Recording critical error 3 at src/main.cpp:635
INFO  | ??:??:?? 10 PowerFSM init, USB power=1
INFO  | ??:??:?? 10 Enter state: BOOT`

closed 2023-05-23 by thebentern