meshtastic/firmware #2301: [Bug]: Heap-Leak at Web-Browser connect

karamo, 2023-02-23:
### Category

Other

### Hardware

T-Beam

### Firmware Version

FW 2.0.21+tryfix

### Description

Normal state... I just made a web browser connect (Chrome) > negative spike > jump ≈+10k ... ... ... no actions made on the web browser, only open > continuous drop to heap reboot ... after that it remains normal, although the web browser is still connected and I can send a Msg.

WZ1 (blue linie) FW 2.0.21+tryfix
magenta & orange are only for reference

[image]

### I need help!
I have no idea where to start troubleshooting.

garthvh, 2023-03-14:
I would start with the response to wantConfig

sachaw, 2023-03-15:
What connection method were you using, can you replicate the behaviour across all 3 methods?

karamo, 2023-03-16:
With FW 2.1.2 + Patch #2329 Heap ist ok.
No longer Heap-Leak.

Issue can bee closed.

karamo, 2023-03-16:
> What connection method were you using, can you replicate the behaviour across all 3 methods?

I connected in Chrome-Browser via IP.
This ist the clearest methode.

closed 2023-03-16 by caveman99