meshtastic/firmware #2307: [Bug]:

Josh-SMU, 2023-02-24:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.0.17

### Description

My devices don't always show traffic in serial debugging, after a non-determinate amount of reboots it works as expected. Troubleshooting and logs in #2291

GUVWAF, 2023-02-24:
If it's only intermittently, this might just be related to a low-level issue I found with SX1262 radios that are dropping packets if they are switching to channel activity detection mode when actively receiving. A change in the library RadioLib where Meshtastic depends on is required for that. 

If you don't mind I will hijack this issue for that while I'm actively going after this.

Josh-SMU, 2023-02-24:
Please do!

This makes sense as I also notice this issue (at least of intermittenly failing to receive messages without other explanation) on the SX1262 chip in the Pinephone backplate project I'm currently writing.

PR #2333 (2023-03-06): SX126x/8x: Add HEADER_VALID IRQ flag for actively receiving check
Fixes #2307.

This updates the RadioLib dependency to the latest release which includes user-configurable IRQ flags for SX126x and SX128x radios. Now the `isActivelyReceiving()` function actually works, which gives way better reliability.
closed 2023-03-06 by thebentern