meshtastic/firmware #2313: since upgrade from 2.0.20 to 2.0.21no contact, preamble change the cause?

Marc12O, 2023-02-27:
So when I upgrade my 2.0.20 node to 2.0.21, I can not communicate anymore to my peers with 2.0.20, they seem not to hear me, when I watch on waterfall display of my SDR receiver, the transmission looks much shorter than before...

So if I say that 2.0.20 node does not recognize 2.0.21 node, does that make sense?

Thanks for the fantastic mesh system by the way!

garthvh, 2023-02-27:
The preamble is now 16 instead of 32 so yes, half the size.

I have a mixed mesh here with 2.0.18 2.0.20 and 2.0.22 and they are all communicating on 915

Marc12O, 2023-02-27:
OK, thank you, I will keep on trying...

mc-hamster, 2023-02-27:
@garthvh @thebentern did we change the sleep timeouts along with the preamble change?

Marc12O, 2023-02-28:
I tried a lot of things, upgrading, clean install with erase, on 2.0.21 2.0.22 2.0.23, no success... could it be frequency (chip) related?

Or maybe there could be another reason? I saw there is also an extra DIO pin change from 2.0.20 to 2.0.21
(Use LORA_DIO1 as RadioLib GPIO for SX127x chips)
https://github.com/meshtastic/firmware/commit/95bbcd2cb7411c10e34747c1f2c65da30daacecf

My board probably does not have a wired DIO1

Thanks for your thoughts!

GUVWAF, 2023-02-28:
@mc-hamster SX1276 radios (like the T-LoRa has) are in Rx continuous mode. For SX126x radios, the sleep intervals are automatically assigned based on the preamble length by the function `startReceiveDutyCycleAuto()`.

@Marc12O There was a long-standing bug for SX1276 radios where although the preamble was set to 32, it actually got set to 100. That's why you see such a large difference in transmission length (which results into much lower airtime utilization). 
If your board doesn't have the DIO1 wired up, there is actually no change. @thebentern tested this change on several devices.
He and I also tested that communicating with mixed preamble lengths (either using SX1276 or SX1262) works. 
Can you please double check (e.g. with `meshtastic --info`) whether you set the channel correctly? Otherwise, we can check whether only changing the preamble length works for you.

Marc12O, 2023-02-28:
Hi, I tried changing the preamble to 32 and 64, even to 8 in de 2.0.23 source, to see if there was a difference, but that did not help, setting up vscode with platformio went surprisingly easy :-)

I watched the QR code in the Android app, it stayed the same, but I will check again

Thanks!

Marc12O, 2023-02-28:
Checked channel keys via meshtasic CLI, and they are the same on all versions I tried... but no succes with firmwares > 2.0.20 in communicating with = 2.0.20 on the other side, on EU_433 with tlora-v2-1-1.6 boards

As soon as I build and upload 2.0.20 it just works instantly

GUVWAF, 2023-02-28:
Can you try by changing `#define LORA_DIO1 33` into `#define LORA_DIO1 RADIOLIB_NC` in _variants/tlora_v2_1_16/variant.h_?

Marc12O, 2023-02-28:
Yes, I tried, no success, after that also DIO2 to RADIOLIB_NC no success, and also preambleLength to 32 in RadioInterface.h , no success...

With 20.0.23

GUVWAF, 2023-02-28:
Can you receive packets from your other nodes when on 2.0.23?
It would also be useful to post the output of `meshtastic –info` for the node on 2.0.23 and one on 2.0.20.

Marc12O, 2023-02-28:
`meshtastic –info` for 2.0.20
```
Connected to radio

Owner: Meshtastic 4330 (4330)
My info: { "myNodeNum": 633291568, "maxChannels": 8, "firmwareVersion": "2.0.20", "rebootCount": 2, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 19.535, "airUtilTx": 0.1685 }
Nodes in mesh:  {'num': 633291568, 'user': {'id': '!25bf4330', 'longName': 'Meshtastic 4330', 'shortName': '4330', 'macaddr': '4c:75:25:bf:43:30', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523548983, 'longitudeI': 49034716, 'altitude': -25, 'time': 1677594453, 'seqNumber': 1, 'latitude': 52.354898299999995, 'longitude': 4.9034716}, 'lastHeard': 1677594453, 'deviceMetrics': {}}  {'num': 633286280, 'user': {'id': '!25bf2e88', 'longName': 'Meshtastic 2e88', 'shortName': '2e88', 'macaddr': '4c:75:25:bf:2e:88', 'hwModel': 'TLORA_V2_1_1P6'}, 'snr': 1.75, 'lastHeard': 1677594432}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_433", "hopLimit": 3, "txEnabled": true, "txPower": 12 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "...." }

Primary channel URL: https://meshtastic.org/e/#....

```

Marc12O, 2023-02-28:
`meshtastic –info` for 2.0.23
```
Connected to radio

Owner: Meshtastic 4330 (4330)
My info: { "myNodeNum": 633291568, "maxChannels": 8, "firmwareVersion": "2.0.23.408c555f", "rebootCount": 3, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 6.57, "airUtilTx": 0.24561112 }
Nodes in mesh:  {'num': 633291568, 'user': {'id': '!25bf4330', 'longName': 'Meshtastic 4330', 'shortName': '4330', 'macaddr': '4c:75:25:bf:43:30', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523549266, 'longitudeI': 49035666, 'altitude': -28, 'time': 1677594918, 'latitude': 52.3549266, 'longitude': 4.9035665999999996}, 'lastHeard': 1677594918, 'deviceMetrics': {'voltage': 4.555, 'channelUtilization': 1.5583334, 'airUtilTx': 0.13611111}}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_433", "hopLimit": 3, "txEnabled": true, "txPower": 12 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "...." }

Primary channel URL: https://meshtastic.org/e/#....

```

Marc12O, 2023-02-28:
Till now I did not receive anything from the other node = 2.0.20  when I am on firmware > 2.0.20, I will leave the board on for a while, the peer is switched on, and should advertise itself also I guess?

Marc12O, 2023-02-28:
The peer list updated, so it seems some receiving occurs here on 2.0.23, or is that via the android app on both sides?

```
Owner: Meshtastic 4330 (4330)
My info: { "myNodeNum": 633291568, "maxChannels": 8, "firmwareVersion": "2.0.23.408c555f", "rebootCount": 5, "bitrate": 59.622643, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 7.653333, "airUtilTx": 0.7352222 }
Nodes in mesh:  {'num': 633291568, 'user': {'id': '!25bf4330', 'longName': 'Meshtastic 4330', 'shortName': '4330', 'macaddr': '4c:75:25:bf:43:30', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523548299, 'longitudeI': 49035116, 'altitude': -19, 'time': 1677596719, 'seqNumber': 5, 'latitude': 52.3548299, 'longitude': 4.9035116}, 'lastHeard': 1677596719, 'deviceMetrics': {'voltage': 4.542, 'channelUtilization': 2.87, 'airUtilTx': 0.6076667}}  {'num': 633286280, 'user': {'id': '!25bf2e88', 'longName': 'Meshtastic 2e88', 'shortName': '2e88', 'macaddr': '4c:75:25:bf:2e:88', 'hwModel': 'TLORA_V2_1_1P6'}, 'snr': 2.75, 'lastHeard': 1677596465, 'deviceMetrics': {'batteryLevel': 87, 'voltage': 4.086, 'airUtilTx': 1.5903611}}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_433", "hopLimit": 3, "txEnabled": true, "txPower": 12 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "AQ==" }

Primary channel URL: https://meshtastic.org/e/#CgMSAQESCggBOAJAA0gBUAw
```

Marc12O, 2023-02-28:
The peer list updated, so it seems some receiving occurs here on 2.0.23, or is that via the android app on both sides?
```

Owner: Meshtastic 4330 (4330)
My info: { "myNodeNum": 633291568, "maxChannels": 8, "firmwareVersion": "2.0.23.408c555f", "rebootCount": 5, "bitrate": 59.622643, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 7.653333, "airUtilTx": 0.7352222 }
Nodes in mesh:  {'num': 633291568, 'user': {'id': '!25bf4330', 'longName': 'Meshtastic 4330', 'shortName': '4330', 'macaddr': '4c:75:25:bf:43:30', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523548299, 'longitudeI': 49035116, 'altitude': -19, 'time': 1677596719, 'seqNumber': 5, 'latitude': 52.3548299, 'longitude': 4.9035116}, 'lastHeard': 1677596719, 'deviceMetrics': {'voltage': 4.542, 'channelUtilization': 2.87, 'airUtilTx': 0.6076667}}  {'num': 633286280, 'user': {'id': '!25bf2e88', 'longName': 'Meshtastic 2e88', 'shortName': '2e88', 'macaddr': '4c:75:25:bf:2e:88', 'hwModel': 'TLORA_V2_1_1P6'}, 'snr': 2.75, 'lastHeard': 1677596465, 'deviceMetrics': {'batteryLevel': 87, 'voltage': 4.086, 'airUtilTx': 1.5903611}}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_433", "hopLimit": 3, "txEnabled": true, "txPower": 12 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "...." }

Primary channel URL: https://meshtastic.org/e/#....

```

garthvh, 2023-02-28:
That info is from the device nodedb, so they communicated.

Marc12O, 2023-02-28:
It is nice that the devices communicate in some way, but the human beings would like to do so too ;-)

I still get little clouds with nasty strikes in the meshtastic chat, no acknowledges

Marc12O, 2023-02-28:
I have also lying around 2 boards 868 MHz, from the same brand and the same format and the same firmware type, there I have the 2.0.20 and 2.0.22 firmwares running, they are both physical here, so I can provide you the output of the `meshtastic --info` command from both nodes

```
$ bin/meshtastic --info --port /dev/ttyACM0
Connected to radio

Owner: M868_f410 (M868)
My info: { "myNodeNum": 633533456, "maxChannels": 8, "firmwareVersion": "2.0.22.fbfd0f1", "rebootCount": 18, "bitrate": 77.526985, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 6.9933333, "airUtilTx": 0.2924722 }
Nodes in mesh:  {'num': 633533456, 'user': {'id': '!25c2f410', 'longName': 'M868_f410', 'shortName': 'M868', 'macaddr': '4c:75:25:c2:f4:10', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523548933, 'longitudeI': 49034750, 'altitude': 6, 'time': 1677627396, 'seqNumber': 1, 'latitude': 52.3548933, 'longitude': 4.903474999999999}, 'lastHeard': 1677627396, 'deviceMetrics': {'voltage': 4.27, 'airUtilTx': 0.19097222}}

Preferences: { "device": { "serialEnabled": true, "nodeInfoBroadcastSecs": 10800 }, "position": { "positionBroadcastSecs": 900, "fixedPosition": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600, "autoScreenCarouselSecs": 30 }, "lora": { "usePreset": true, "region": "EU_868", "hopLimit": 3, "txEnabled": true, "txPower": 27 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "...." }

Primary channel URL: https://meshtastic.org/e/#....

$ bin/meshtastic --info --port /dev/ttyACM1
Connected to radio

Owner: M868_ee94 (M868)
My info: { "myNodeNum": 633532052, "maxChannels": 8, "firmwareVersion": "2.0.20.7100416", "rebootCount": 9, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "hasWifi": true, "channelUtilization": 13.666667, "airUtilTx": 0.3568889 }
Nodes in mesh:  {'num': 633532052, 'user': {'id': '!25c2ee94', 'longName': 'M868_ee94', 'shortName': 'M868', 'macaddr': '4c:75:25:c2:ee:94', 'hwModel': 'TLORA_V2_1_1P6'}, 'position': {'latitudeI': 523548683, 'longitudeI': 49034733, 'altitude': 15, 'time': 1677627364, 'seqNumber': 1, 'latitude': 52.3548683, 'longitude': 4.9034733}, 'lastHeard': 1677627364, 'deviceMetrics': {'batteryLevel': 64, 'voltage': 3.869, 'channelUtilization': 1.1083333, 'airUtilTx': 0.16672222}}

Preferences: { "device": { "serialEnabled": true }, "position": { "positionBroadcastSecs": 900, "positionBroadcastSmartEnabled": true, "gpsEnabled": true, "gpsUpdateInterval": 120, "gpsAttemptTime": 900, "positionFlags": 3, "rxGpio": 15, "txGpio": 13 }, "power": { "waitBluetoothSecs": 60, "meshSdsTimeoutSecs": 7200, "sdsSecs": 4294967295, "lsSecs": 300, "minWakeSecs": 10 }, "network": { "ntpServer": "0.pool.ntp.org" }, "display": { "screenOnSecs": 600 }, "lora": { "usePreset": true, "region": "EU_868", "hopLimit": 3, "txEnabled": true, "txPower": 27 }, "bluetooth": { "enabled": true, "fixedPin": 123456 } }

Module preferences: { "mqtt": { "address": "mqtt.meshtastic.org", "username": "meshdev", "password": "large4cats" }, "serial": {}, "externalNotification": {}, "rangeTest": {}, "telemetry": { "deviceUpdateInterval": 900, "environmentUpdateInterval": 900 }, "cannedMessage": {} }

Channels:
  PRIMARY psk=default { "psk": "...." }

Primary channel URL: https://meshtastic.org/e/#....
```

Channel keys were the same!

GUVWAF, 2023-03-01:
I don't see any obvious misconfigurations here. Which modem preset are you using?
Can you try with the default channel on both of these?

Marc12O, 2023-03-01:
I thought I was already using that, the LongFast preset, works fine with 2.0.20 <-> 2.0.20

Marc12O, 2023-03-01:
The good news is, that when I upgrade both EU_868 boards, to 2.0.23 <-> 2.0.23, all works fine :-)

Marc12O, 2023-03-01:
Just for completeness and maybe more, here 2 pictures of my 433 board

[image]

[image]

GUVWAF, 2023-03-01:
Thanks for all the info. I am not sure why it doesn't work between 2.0.20 and higher, but at least it is good to see when they're both updated it works.
Version 2.0.20 has been revoked now and 2.0.23 is released. If 2.0.23 also works on your 433MHz modules, I suggest we can close this?