meshtastic/firmware #2318: [Bug]: SX1262 radios crash by receiving wron packet

karamo, 2023-03-01:
### Category

Other

### Hardware

T-Beam

### Firmware Version

FW 2.0.21+tryfix

### Description

By receiving a packet, after decoding "wrong" portnum & payloadlen:

```
DEBUG | 04:13:57 839 [RadioIf] Lora RX (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=6 Ch=0x9b encrypted rxSNR=0.5 rxRSSI=-102AirTime - Packet received : 410ms
DEBUG | 04:13:57 839 [RadioIf] ------ Thread RadioIf leaked heap 49112 -> 48788 (-324) ------
DEBUG | 04:13:57 839 [Router] Add packet record (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=6 Ch=0x9b encrypted rxSNR=0.5 rxRSSI=-102No suitable channel found for decoding, hash was 0x9b!
DEBUG | 04:13:57 839 [Router] packet decoding failed or skipped (no PSK?) (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=6 Ch=0x9b encrypted rxtime=1677471237 rxSNR=0.5 rxRSSI=-102Module 'routing' wantsPacket=1

INFO  | 04:13:57 839 [Router] Received routing from=0xf246b4e4, id=0x35b6596c, portnum=-1841823716, payloadlen=15965

DEBUG | 04:13:57 839 [Router] Routing sniffing (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=6 Ch=0x9b encrypted rxtime=1677471237 rxSNR=0.5 rxRSSI=-102Rebroadcasting received floodmsg to neighbors
DEBUG | 04:13:57 839 [Router] enqueuing for send (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=5 Ch=0x9b encrypted rxtime=1677471237 rxSNR=0.5 rxRSSI=-102txGood=44,rxGood=99,rxBad=25
DEBUG | 04:13:57 839 [Router] rx_snr found. hop_limit:5 rx_snr:0.500000
DEBUG | 04:13:57 839 [Router] rx_snr found in packet. Setting tx delay:126
DEBUG | 04:13:57 839 [Router] Delivering rx packet (id=0x35b6596c fr=0xe4 to=0xff, WantAck=0, HopLim=6 Ch=0x9b encrypted rxtime=1677471237 rxSNR=0.5 rxRSSI=-102Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

```
**Backtrace:**
    0x400e5c03: MeshService::handleFromRadio(_meshtastic_MeshPacket const*) at \src/mesh\MeshService.cpp:79
    0x400f5a25: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at \src/modules\RoutingModule.cpp:19
    0x400f5a25: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at \src/modules\RoutingModule.cpp:10
    0x400f59a5: ProtobufModule<_meshtastic_Routing>::handleReceived(_meshtastic_MeshPacket const&) at \src/mesh\ProtobufModule.h:87
    0x400e5056: MeshModule::callPlugins(_meshtastic_MeshPacket const&, RxSource) at \src/mesh\MeshModule.cpp:128
    0x400ea411: Router::handleReceived(_meshtastic_MeshPacket*, RxSource) at \src/mesh\Router.cpp:461
    0x400ea4d5: Router::perhapsHandleReceived(_meshtastic_MeshPacket*) at \src/mesh\Router.cpp:478
    0x400ea4f1: Router::runOnce() at \src/mesh\Router.cpp:69
    0x400e9c09: ReliableRouter::runOnce() at \src/mesh\ReliableRouter.h:79
    0x400d981b: concurrency::OSThread::run() at \src/concurrency\OSThread.cpp:81
    0x401025dd: ThreadController::runOrDelay() at \.pio/libdeps/tbeam/Thread\ThreadController.cpp:59
    0x400e299c: loop() at \src\main.cpp:587
    0x40139921: loopTask(void*) at /.platformio/packages/framework-arduinoespressif32/cores/esp32\main.cpp:50

GUVWAF, 2023-03-02:
Thanks for reporting this with the logs and backtrace. I think I know why this goes wrong and will try to fix this soon.

closed 2023-03-02 by thebentern

mverch67, 2023-03-03:
I've added a comment to the PR #2320 . There may be a new bug unrelated to the original wrong packet receive.

PR #2320 (2023-03-02): Check if packet is decrypted before searching node in DB
Fixes #2318. Only happened when you got a message you could decrypt and from a node you did not yet have in the nodeDB.

Also added a check for the TraceRouteModule, in case it will not be enabled it for some role in the future.