meshtastic/firmware #2319: [Bug]: "Tracker" device profile does not honor Smart Positioning

djscoutmaster, 2023-03-02:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.21

### Description

**Problem:** 
When a device profile is set to "tracker" it does not honor the smart positioning flag feature. It continues to report location packets even after stopped.

**Steps to reproduce**
1. Download 2.0.21.beta onto TBeam
2. Connect via bluetooth on iPhone app
3. Navigate to Settings > Device > Device role = Tracker > Save Settings
4. Wait for device to reboot
5. Navigate to Settings > Position > Smart Broadcast = Enabled & Position Broadcast Interval = 15 seconds
6. Wait for device to reboot
7. Start driving in a car
8. Bring car to stop
9. Device continues to report location ever 15 seconds and does not stop

thebentern, 2023-03-02:
Can you run a meshtastic --info and confirm that smart broadcast is enabled? I'm suspecting it might not be. Setting to Tracker role turns it off by default but doesn't prevent you from setting it back to enabled.

djscoutmaster, 2023-03-11:
I have confirmed this on many devices... on all my teams when smart positioning (positionBroadcastSmartEnabled) is true and tracker is true... it does not stop reporting when I com to a stop... it continues reporting on 15 sec intervals

Preferences: { "device": { "role": "TRACKER", "serialEnabled": true, "nodeInfoBroadcastSecs": 10800 }, "position": { "positionBroadcastSecs": 15, "positionBroadcastSmartEnabled": true...

garthvh, 2023-03-16:
I think we need to look at the timing of the smart positioning, it appears that the settings are all configured properly.

djscoutmaster, 2023-03-18:
@garthvh  and @thebentern the issue im concerned about is that when you come to a stop... the position timing set (15sec in this example) continues even if you have not moved... the will congest the network in if its doesnt honor the rule that requires both time and distance to be met in order to broadcast a new position

GUVWAF, 2023-03-25:
> Really the issue is that you should be able to set the interval to 15 minutes or an hour and get more frequent updates than that if the node moves, I don't think this is how it is currently working.

I think this is how it is currently working, but @djscoutmaster has the interval set to 15 seconds, meaning that it will send _at least_ every 15 seconds, but _also_ if you moved enough (Smart Positioning). Quoted from the [docs](https://meshtastic.org/docs/settings/config/position#smart-broadcast):

> Smart broadcast complements broadcast interval (doesn't override that setting) but will apply an algorithm to more frequently update your mesh network if you are in motion and then throttle it down when you are standing still. If you use this feature, it's best to leave broadcast interval at the default.

If you're mainly interested in position packets when you move, I believe you should enable smart positioning and set the `positionBroadcastSecs` higher.

djscoutmaster, 2023-03-29:
With the new changes to smart positioning this is no longer and issue

closed 2023-03-29 by djscoutmaster