meshtastic/firmware #2321: [Bug]: 3x repeated messages in REPEATER mode

mverch67, 2023-03-03:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.0.23.7bb281d

### Description

1. set device.role REPEATER
2. connect with Android APP
3. Send a text message in any available channel

Result: the APP is repeating the text message three times (every 45s) and the same message is sent out three times by the device. It seems the device in role REPEATER is not sending an ACK back to the client as it does in other roles. Thus, the APP repeats all text messages in all channels where it didn't receive an ACK so far. It does this with every BLE reconnect. Sometimes it repeats forever (which is another bug). 

[image]

### Relevant log output

```Shell
device.role = REPEATER:

DEBUG | ??:??:?? 60 PACKET FROM PHONE (id=0x252ed3ca fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x1 Portnum=1handleReceived(USER) (id=0x252ed3ca fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x1 Portnum=1Module 'routing' wantsPacket=1
INFO  | ??:??:?? 60 Received routing from=0x0, id=0x252ed3ca, portnum=1, payloadlen=3
DEBUG | ??:??:?? 60 Routing sniffing (id=0x252ed3ca fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x1 Portnum=1Module 'routing' considered
DEBUG | ??:??:?? 60 Add packet record (id=0x252ed3ca fr=0x00 to=0xff, WantAck=1, HopLim=0 Ch=0x1 Portnum=1Original length - 3 
DEBUG | ??:??:?? 60 Compressed length - 4 
DEBUG | ??:??:?? 60 Original message - Bug 
DEBUG | ??:??:?? 60 Not using compressing message.
DEBUG | ??:??:?? 60 Using AES256 key!
DEBUG | ??:??:?? 60 ESP32 crypt fr=6d088430, num=252ed3ca, numBytes=7!
DEBUG | ??:??:?? 60 enqueuing for send (id=0x252ed3ca fr=0x30 to=0xff, WantAck=0, HopLim=0 Ch=0x9d encryptedtxGood=0,rxGood=0,rxBad=0
DEBUG | ??:??:?? 60 [RadioIf] Starting low level send (id=0x252ed3ca fr=0x30 to=0xff, WantAck=0, HopLim=0 Ch=0x9d encrypted priority=64(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=23, time 2105 ms
DEBUG | ??:??:?? 60 [RadioIf] AirTime - Packet transmitted : 2105ms
INFO  | ??:??:?? 60 Telling client we have new packets 2
INFO  | ??:??:?? 60 BLE notify fromNum
INFO  | ??:??:?? 60 From Radio onread
INFO  | ??:??:?? 60 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 60 encoding toPhone packet to phone variant=11, 12 bytes
INFO  | ??:??:?? 60 From Radio onread
DEBUG | ??:??:?? 62 [Power] Battery: usbPower=0, isCharging=0, batMv=4188, batPct=98
DEBUG | ??:??:?? 62 [RadioIf] Completed sending (id=0x252ed3ca fr=0x30 to=0xff, WantAck=0, HopLim=0 Ch=0x9d encrypted priority=64(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=63, time 4202 ms
DEBUG | ??:??:?? 70 [RadioIf] Lora RX (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=3 Ch=0x2f encrypted rxSNR=7.75 rxRSSI=-73AirTime - Packet received : 4202ms
DEBUG | ??:??:?? 70 [Router] Add packet record (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=3 Ch=0x2f encrypted rxSNR=7.75 rxRSSI=-73Using channel 0 (hash 0x2f)
DEBUG | ??:??:?? 70 [Router] Using AES256 key!
DEBUG | ??:??:?? 70 [Router] ESP32 crypt fr=7a9ea4e9, num=3885320b, numBytes=47!
DEBUG | ??:??:?? 70 [Router] decoded message (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxSNR=7.75 rxRSSI=-73handleReceived(REMOTE) (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxSNR=7.75 rxRSSI=-73Module 'routing' wantsPacket=1
INFO  | ??:??:?? 70 [Router] Received routing from=0x7a9ea4e9, id=0x3885320b, portnum=4, payloadlen=41
DEBUG | ??:??:?? 70 [Router] Routing sniffing (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxSNR=7.75 rxRSSI=-73Rebroadcasting received floodmsg to neighbors
DEBUG | ??:??:?? 70 [Router] Using AES256 key!
DEBUG | ??:??:?? 70 [Router] ESP32 crypt fr=7a9ea4e9, num=3885320b, numBytes=47!
DEBUG | ??:??:?? 70 [Router] enqueuing for send (id=0x3885320b fr=0xe9 to=0xc8, WantAck=0, HopLim=2 Ch=0x2f encrypted rxSNR=7.75 rxRSSI=-73txGood=1,rxGood=1,rxBad=0
DEBUG | ??:??:?? 70 [Router] rx_snr found. hop_limit:2 rx_snr:7.750000
DEBUG | ??:??:?? 70 [Router] rx_snr found in packet. As a router, setting tx delay:462
DEBUG | ??:??:?? 70 [Router] Module 'routing' considered
WARN  | ??:??:?? 70 [RadioIf] Can not send yet, busyRx
DEBUG | ??:??:?? 70 [RadioIf] rx_snr found. hop_limit:2 rx_snr:7.750000
DEBUG | ??:??:?? 70 [RadioIf] rx_snr found in packet. As a router, setting tx delay:420
WARN  | ??:??:?? 71 [RadioIf] Can not send yet, busyRx
DEBUG | ??:??:?? 71 [RadioIf] rx_snr found. hop_limit:2 rx_snr:7.750000
DEBUG | ??:??:?? 71 [RadioIf] rx_snr found in packet. As a router, setting tx delay:168
WARN  | ??:??:?? 71 [RadioIf] Can not send yet, busyRx
DEBUG | ??:??:?? 71 [RadioIf] rx_snr found. hop_limit:2 rx_snr:7.750000
DEBUG | ??:??:?? 71 [RadioIf] rx_snr found in packet. As a router, setting tx delay:294
WARN  | ??:??:?? 71 [RadioIf] Can not send yet, busyRx
[... 232 lines cut ...]
DEBUG | ??:??:?? 142 [Router] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=23, time 2105 ms
DEBUG | ??:??:?? 142 [Router] Setting next retransmission in 10726 msecs:  (id=0x252ed3c8 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x1 Portnum=1Starting low level send (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=23, time 2105 ms
DEBUG | ??:??:?? 142 [RadioIf] AirTime - Packet transmitted : 2105ms
DEBUG | ??:??:?? 142 [Power] Battery: usbPower=0, isCharging=0, batMv=4181, batPct=97
DEBUG | ??:??:?? 144 [RadioIf] Completed sending (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=3 Ch=0x9d encrypted priority=64(bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=23, time 2105 ms
DEBUG | ??:??:?? 147 [RadioIf] Lora RX (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=2 Ch=0x9d encrypted rxSNR=12.25 rxRSSI=-101AirTime - Packet received : 2105ms
DEBUG | ??:??:?? 147 [Router] Rx someone rebroadcasting for us (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=2 Ch=0x9d encrypted rxSNR=12.25 rxRSSI=-101generating implicit ack
ERROR | ??:??:?? 147 [Router] Alloc an err=0,to=0x6d088430,idFrom=0x252ed3c8,id=0x4d0f3587
DEBUG | ??:??:?? 147 [Router] Enqueued local (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120cancelSending id=0x252ed3c8, removed=0
DEBUG | ??:??:?? 147 [Router] Found existing packet record for fr=0x6d088430,to=0xffffffff,id=0x252ed3c8
DEBUG | ??:??:?? 147 [Router] Found existing packet record for fr=0x6d088430,to=0xffffffff,id=0x252ed3c8
DEBUG | ??:??:?? 147 [Router] Add packet record (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=2 Ch=0x9d encrypted rxSNR=12.25 rxRSSI=-101Ignoring incoming msg, because we've already seen it (id=0x252ed3c8 fr=0x30 to=0xff, WantAck=0, HopLim=2 Ch=0x9d encrypted rxSNR=12.25 rxRSSI=-101Incoming message was filtered 0x6d088430
DEBUG | ??:??:?? 147 [Router] Rx someone rebroadcasting for us (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120didn't find pending packet
DEBUG | ??:??:?? 147 [Router] Add packet record (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120handleReceived(REMOTE) (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120Module 'routing' wantsPacket=1
INFO  | ??:??:?? 147 [Router] Received routing from=0x6d088430, id=0x4d0f3587, portnum=5, payloadlen=2
DEBUG | ??:??:?? 147 [Router] Routing sniffing (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120Received an ack for 0x252ed3c8, stopping retransmissions
DEBUG | ??:??:?? 147 [Router] Delivering rx packet (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120Update DB node 0x6d088430, rx_time=0
DEBUG | ??:??:?? 147 [Router] Forwarding to phone (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120Module 'routing' considered
INFO  | ??:??:?? 147 Telling client we have new packets 10
INFO  | ??:??:?? 147 BLE notify fromNum
INFO  | ??:??:?? 147 From Radio onread
INFO  | ??:??:?? 147 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 147 phone downloaded packet (id=0x4d0f3587 fr=0x30 to=0x30, WantAck=0, HopLim=3 Ch=0x1 Portnum=5 requestId=252ed3c8 priority=120encoding toPhone packet to phone variant=2, 36 bytes
INFO  | ??:??:?? 147 From Radio onread
```

garthvh, 2023-03-03:
Repeater does not have the routing module and does not ack

garthvh, 2023-03-03:
Moving to android, can't replicate with iOS or the CLI