meshtastic/firmware #2322: [Feature Request]: Transmit Only Devices

tas246, 2023-03-04:
### Platform

NRF52, ESP32

### Description

Would love a feature that allows for static mesh setup with 8-10 nodes with hundreds of mostly silent sensors.  The sensors would be distributed and all within range of the mesh nodes, but only transmitting data out occasionally when certain sensors numbers are out of bounds.  The goal would be to have many hundreds of mostly idle sensors and have them sit in a power conserve mode until an issue occurs then transmit the condition.    

The sensor nodes would not receive or retransmit any mesh data as to avoid network congestion. Trying to flag these emergencies to a central control spot.

garthvh, 2023-03-04:
I think the pieces required for this are all in place, the code that processes and reacts to the data from the individual sensor would run on a different microcontroller.

In theory you could use a ESP32 cam with a PIR to detect motion and take some pictures, based on some logic on the esp32 cam you would

* Power on the meshtastic board
* Talk to it over serial and either send messages or telemetry packets
* Send the admin message to shutdown the connected lora radio

thebentern, 2023-05-23:
This functionality seems to already be at least mostly in place with existing configurations