meshtastic/firmware #2328: [Bug]: TBEAM Reboots after some Minutes

daniel-hofer, 2023-03-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.23.7bb281d

### Description

TBeam reboots after some minutes. Changes I made:
Update / Reinstall to firmware 2.0.23.7bb281d.
Attached a USB-Hub on the OTG-Port of the Raspberry Pi Zero together with the Node - so maybe it could be a power-pupply-issue. Will disconnect it and test again. Thank you

[meshtasticinfo.txt](https://github.com/meshtastic/firmware/files/10893064/meshtasticinfo.txt)
[meshtasticlog.txt](https://github.com/meshtastic/firmware/files/10893065/meshtasticlog.txt)

daniel-hofer, 2023-03-05:
OK, tested again with power supply directly attached to the TBeam with the same result of reboots after some minutes.

GUVWAF, 2023-03-05:
I think this is already fixed by #2320. Can you please try with the firmware as included in the zip file of the artifact in that PR?

garthvh, 2023-03-05:
0x400e57a3: MeshService::handleFromRadio(_meshtastic_MeshPacket const*) at /home/runner/work/firmware/firmware/src/mesh/MeshService.cpp:79
0x400f568d: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at /home/runner/work/firmware/firmware/src/modules/RoutingModule.cpp:19
0x400f568d: RoutingModule::handleReceivedProtobuf(_meshtastic_MeshPacket const&, _meshtastic_Routing*) at /home/runner/work/firmware/firmware/src/modules/RoutingModule.cpp:10
0x400f560d: ProtobufModule<_meshtastic_Routing>::handleReceived(_meshtastic_MeshPacket const&) at /home/runner/work/firmware/firmware/src/mesh/ProtobufModule.h:87
0x400e4bf6: MeshModule::callPlugins(_meshtastic_MeshPacket const&, RxSource) at /home/runner/work/firmware/firmware/src/mesh/MeshModule.cpp:128
0x400ea01d: Router::handleReceived(_meshtastic_MeshPacket*, RxSource) at /home/runner/work/firmware/firmware/src/mesh/Router.cpp:461
0x400ea0e1: Router::perhapsHandleReceived(_meshtastic_MeshPacket*) at /home/runner/work/firmware/firmware/src/mesh/Router.cpp:478
0x400ea0fd: Router::runOnce() at /home/runner/work/firmware/firmware/src/mesh/Router.cpp:69
0x400e97fd: ReliableRouter::runOnce() at /home/runner/work/firmware/firmware/src/mesh/ReliableRouter.h:79
0x400d939e: concurrency::OSThread::run() at /home/runner/work/firmware/firmware/src/concurrency/OSThread.cpp:81
0x401022c9: ThreadController::runOrDelay() at /home/runner/work/firmware/firmware/.pio/libdeps/tbeam/Thread/ThreadController.cpp:59
0x400e2548: loop() at /home/runner/work/firmware/firmware/src/main.cpp:587
0x40139639: loopTask(void*) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

GUVWAF, 2023-03-06:
The stack trace is the same as in issue #2318, so highly likely it's already fixed.