meshtastic/firmware #2343: [Bug]: ignore_incoming ignore routed packages, can't be used for testing

khrystynaO, 2023-03-10:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.0.23.7bb281d

### Description

When we set nodenum in config ignore_incoming list  - device ignores all types of packets including routed ones. 

What can we use for debugging routing packages through a third device?

garthvh, 2023-03-10:
Ignore incoming will indeed artificially drop all packets from that node, have you used the trace route function?  That is the only way currently to track packets across nodes.

khrystynaO, 2023-03-13:
What is the correct way to implement this feature?

GUVWAF, 2023-03-20:
How to use the Traceroute is documented [here](https://meshtastic.org/docs/settings/moduleconfig/traceroute). 
Not ignoring routed packets would require breaking changes, because Meshtastic normally does not keep track of the ID of a node that is only rebroadcasting/relaying a message. A routed packet is an exact copy of the original one, except for the hop limit, which is bumped down by one.

closed 2023-05-23 by thebentern