meshtastic/firmware #2344: [Bug]: aes keys are padded

jmgurney, 2023-03-10:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

current

### Description

In looking at the code, it appears that AES keys are zero padded:
https://github.com/meshtastic/firmware/blob/master/src/mesh/Channels.cpp#L129

This is likely a bad idea and may open up the protocol to future attacks.  I'd recommend to require a full 128 or 256 bit key, but if shorter (or longer) keys are required, then it's highly recommended to use a KDF (https://en.wikipedia.org/wiki/Key_derivation_function ) such a PBKDF2 to expand the shorter (or longer) key to the correct size.  Using something like this would also the use phrases (say any UTF-8 string) as the shared key.

caveman99, 2023-03-11:
Running PBKDF2 in a remotely secure manner exceeds the capabilities of an embedded processor by far. Even SSL Key generation takes a good few seconds as it is. Other KDF's are even more complex than PBKDF2.

Good practise for a user is to use the correct key size, and they usually do. This code is just a safety net for an irresponsible user, so the following code is happy and does not bomb. Most of our keys are random generated anyway, we hardly have users supplying keys manually.

From a 'white collar cryptography' approach i am fully with you, this is not a good practise.

closed 2023-03-11 by caveman99

jmgurney, 2023-03-12:
The iteration count can be adjusted in PBKDF2 to make performance not an issue, plus it's only done once at key load time, so taking a few seconds at that time is fine.  And there more simple KDFs, such as a simple usage of SHAKE256.

Talking w/ a cryptographer friend of mine, he says that zero bytes are void warranty but likely to be fine, though he did say there are related key attacks on AES-256 which may be a problem.

jmgurney, 2023-03-16:
There are related key attacks that weaken AES.  e.g.:
https://link.springer.com/chapter/10.1007/978-3-642-03356-8_14

Key selection is assumed to be uniformly random and cryptanalysis is done based upon that, and when that contract is broken, cracks form.

Yes, it's still strong enough, and CTR is a bigger problem, but every link in the cryptography chain needs to be strong.  Attacks only get better over time.