meshtastic/firmware #2347: [Feature Request]: Create DIY variant.h / pin definitions for ESP32-S3

sgofferj, 2023-03-11:
### Platform

ESP32

### Description

The ESP32-S3 is quite a bit different from the ESP32-noS's. Some of the pins in the DIY variant.h for ESP32 can't be used.

[Espressif documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gpio.html):
>SPI0/1: GPIO26-32 are usually used for SPI flash and PSRAM and not recommended for other uses. When using Octal Flash or Octal PSRAM or both, GPIO33~37 are connected to SPIIO4 - SPIIO7 and SPIDQS. Therefore, on boards embedded with ESP32-S3R8 / ESP32-S3R8V chip, GPIO33-37 are also not recommended for other uses.

I'm working on a board with an ESP32-S3 at the moment . While I am using a variant.h from another project which also uses an ESP32-S3, I'd feel more comfortable, if there was some "official" guidance on the pin use for the S3.

sgofferj, 2023-03-28:
I'm designing my own with an ESP32-S3-WROOM-U and an ebyte E900.

markbirss, 2023-03-29:
@sgofferj here is some guidance on DIY Targets i added
https://github.com/meshtastic/firmware/pull/2391

sgofferj, 2023-03-29:
Thanks, I'll look into. I opened the issue because I wanted to avoid having to build my own version of Meshtastic. Whipping up a board design is task enough for me :rofl:

caveman99, 2023-04-02:
DIY targets added

closed 2023-04-02 by caveman99