meshtastic/firmware #2350: Update with the app

Commander-Crash, 2023-03-11:
### Platform

ESP32

### Description

I have a tbeam that has the unfortunate micro usb broken off. I have updated via the app in the past but now i can not. I would love to see this feature back for devices that have no USB but still work.

rpsainio, 2023-03-11:
TBeam exposes the TX/RX-pins so it is possible to bypass the USB-UART-combo, however I am not  sure if the boot-pin is also exposed so that ESP32 starts in bootmode - have to check that from original docus

garthvh, 2023-03-11:
The firmware part of the ota is in place for both esp32 and nrf

thebentern, 2023-03-12:
I did a write-up a while back about using an external usb to serial adapter to flash a T-Beam if that's the route you want to go down
https://meshtastic.org/docs/getting-started/flashing-firmware/esp32/external-serial-adapter

garthvh, 2023-03-15:
Everything necessary to do this is in place in the firmware

closed 2023-03-15 by garthvh