meshtastic/firmware #2351: [Bug]: Wifi Connection / disconnection loop

iz1kga, 2023-03-11:
### Category

WiFi

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.1.0.331a1af

### Description

The device disconnect and reconnect from wifi, this happens on TBEAM and TLORA board.

No problem detected with FW 2.0.21

Regards

### Relevant log output

```Shell
LOG #1

[748356][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[748356][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
[748359][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 20:51:36 748 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 20:51:36 748 [WebServerThread] WiFi lost connection. Reason: 201
INFO  | 20:51:37 749 [WifiConnect] Reconnecting to WiFi access point IZ1KGA_Router
DEBUG | 20:51:42 754 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=37, time 197 ms
ERROR | 20:51:42 754 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:51:42 754 [RadioIf] AirTime - Packet received (noise?) : 197ms
DEBUG | 20:51:44 756 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=37, time 197 ms
ERROR | 20:51:44 756 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:51:44 756 [RadioIf] AirTime - Packet received (noise?) : 197ms
INFO  | 20:51:44 756 [PowerFSM] lsEnter begin, ls_secs=86400
INFO  | 20:51:44 756 [PowerFSM] GPS prepare sleep!
INFO  | 20:52:14 786 [PowerFSM] GPS prepare sleep!
DEBUG | 20:52:14 786 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
INFO  | 20:52:14 786 [PowerFSM] GPS prepare sleep!
INFO  | 20:52:22 793 [PowerFSM] Exit light sleep gpio: btn=0
INFO  | 20:52:22 793 [PowerFSM] wakeCause2 7
INFO  | 20:52:22 793 [PowerFSM] Exit state: LS
DEBUG | 20:52:22 793 [PowerFSM] Allowing GPS lock
INFO  | 20:52:22 793 [PowerFSM] Enter state: NB
DEBUG | 20:52:22 793 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=40, time 207 ms
ERROR | 20:52:22 793 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:52:22 793 [RadioIf] AirTime - Packet received (noise?) : 207ms
[794211][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[794212][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
[794215][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 20:52:22 794 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 20:52:22 794 WiFi lost connection. Reason: 201
INFO  | 20:52:23 795 [WifiConnect] Reconnecting to WiFi access point IZ1KGA_Router
DEBUG | 20:52:28 800 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=30, time 172 ms
ERROR | 20:52:28 800 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:52:28 800 [RadioIf] AirTime - Packet received (noise?) : 172ms
[801068][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 20:52:29 801 WiFi-Event 4: Connected to access point
[801678][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[801678][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 192.168.1.106, MASK: 255.255.255.0, GW: 192.168.1.1
DEBUG | 20:52:30 801 WiFi-Event 7: Obtained IP address: lsEnter begin, ls_secs=86400
INFO  | 20:52:32 803 [PowerFSM] GPS prepare sleep!
INFO  | 20:53:02 833 [PowerFSM] GPS prepare sleep!
DEBUG | 20:53:02 833 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
INFO  | 20:53:02 834 [PowerFSM] GPS prepare sleep!
INFO  | 20:53:32 864 [PowerFSM] GPS prepare sleep!
DEBUG | 20:53:32 864 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
[... 93 lines cut ...]
INFO  | 20:56:14 1026 [PowerFSM] Exit state: LS
DEBUG | 20:56:14 1026 [PowerFSM] Allowing GPS lock
INFO  | 20:56:14 1026 [PowerFSM] Enter state: NB
DEBUG | 20:56:14 1026 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=38, time 201 ms
ERROR | 20:56:14 1026 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:56:14 1026 [RadioIf] AirTime - Packet received (noise?) : 201ms
DEBUG | 20:56:14 1026 [RadioIf] (bw=250, sf=9, cr=4/8) packet symLen=2 ms, payloadSize=71, time 320 ms
ERROR | 20:56:14 1026 [RadioIf] ignoring received packet due to error=-7
DEBUG | 20:56:14 1026 [RadioIf] AirTime - Packet received (noise?) : 320ms
[1028179][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1028179][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
[1028183][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 20:56:16 1028 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 20:56:16 1028 [WebServerThread] WiFi lost connection. Reason: 201
INFO  | 20:56:17 1029 [WifiConnect] Reconnecting to WiFi access point IZ1KGA_Router

LOG #2
ERROR | 21:51:20 115 [mqtt] Failed to contact MQTT server (5/5)... 
INFO  | 21[115636][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 
[115637][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE 
:51:20 115 [WifiConnect] Reconnecting to WiFi access point TRN 
WARN  | 21:51:20 115 [WifiConnect] WiFi lost connection. Reason: 8

From here loop begins, this is not a log of mine so i cant provide more details
```

garthvh, 2023-03-11:
201 is no ap found

iz1kga, 2023-03-12:
> 201 is no ap found

oh, thanks, maybe my AP hide itself when I use FW version 2.1.x

such a nice A.I.

ik1jns, 2023-03-12:
I've noticed the same behavior Wifi and MQTT enabled after some time the board began to connect and disconnect in a loop 
[image]
[image]

rpsainio, 2023-03-12:
Is there any debugging available to see that WIFI is shutdown before going LS
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html
According to above mentioned is must be done programmatically - so either Arduino or Meshtastic

vinloren, 2023-03-13:
[swcausedconnabort_1040130223.txt](https://github.com/meshtastic/firmware/files/10957955/swcausedconnabort_1040130223.txt)
Here is another example of connection lost and autoreboot as it can be seen at line 226 (Tlora2 fmwr 2.1.0 mqtt connection WiFi active).

vinloren, 2023-03-13:
[image]

closed 2023-08-14 by garthvh