meshtastic/firmware #2356: [Bug]: eink updates causing packet loss

andrekir, 2023-03-13:
### Category

Other

### Hardware

Other

### Firmware Version

2.1.0+

### Description

every packet transmit and receive triggers the e-ink screen to redraw, with the log message ["Updating E-Paper..."](https://github.com/meshtastic/firmware/blob/d92a003d8e8cc5d28728c0299f0f49b4ee297a06/src/graphics/EInkDisplay2.cpp#L107) and looks like nRF52s can't handle well both at the same time.

@caveman99 suggested moving e-ink updates to their own RTOS task.

nickpascucci, 2023-08-21:
Following on from discussion here: https://discord.com/channels/867578229534359593/871553168369148024/1143163122396450928

I flashed my T-Echo with a build at b47c9c16 with a change on line 35 of `src/graphics/EInkDisplay2.h`:

```
bool forceDisplay(uint32_t msecLimit = 100000);
```

This prevented the display from updating as frequently, and I was able to send 26 messages (A-Z) to a nearby T-Beam with no missed acknowledgments. Previously I would have more misses than acks. This seems to strongly point to the e-ink updates being an issue in a hot loop.

GUVWAF, 2023-08-29:
Added the high-priority label as it seems to cause ACK failures quite easily, because transmitting triggers the touch button causing multiple e-ink updates. 
Conclusion of a discussion on Discord was that the missed packets are caused by the screen update blocking ISR access. Therefore, that needs to happen in a dedicated RTOS task.

marek22k, 2023-12-06:
Is there a solution or a workaround without having to compile the firmware yourself?

GUVWAF, 2024-02-10:
I think for the T-Echo this is not an issue anymore now that it has partial updates which also causes it not to trigger the touch button anymore.

I can confirm it's still an issue with the Heltec Wireless Paper which only has full updates.

closed 2024-03-10 by garthvh