meshtastic/firmware #2360: [Bug]: Only send NodeInfo in response to a position request, not on the position schedule

garthvh, 2023-03-16:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.1.2

### Description

When a position is sent directly to another node and a response is requested we send a NodeInfo if the device does not have a position.  This works pretty well and at least lets the other user know that the device responded.

We also seem to send an empty NodeInfo on the position schedule if the device has no GPS, and no fixed or phone provided positions which seems unnecessary.

GUVWAF, 2023-03-25:
I was wondering why I couldn’t replicate this, but I found that it is only sending this to the node that is currently displayed on the screen (if you have one). In fact, calling this `sendNetworkPing` function is done when you enter the power “ON” state and you sent the last ping more than one minute ago.
I agree that this is unnecessary. I also wonder if sending your position to the node that is displayed is even something that is desired?

garthvh, 2023-03-25:
> I was wondering why I couldn’t replicate this, but I found that it is only sending this to the node that is currently displayed on the screen (if you have one). In fact, calling this [...]

Seems undesirable. I only found this because empty position packets crashed the app at one point so I log them.

PR #2387 (2023-03-26): Remove sending network ping to displayed node on interval
Fixes #2360.
This removes sending a network ping on an interval to the node that is currently displayed on a screen. This feature silently overrules the position broadcast setting, or sends its NodeInfo over and over again, which seems unnecessary. It will also create a lot of packets when you set the screen to carousel.
The ‘last seen’ time already gets updated with any other packet received from the node. Besides, this feature is not documented, so I don’t think anyone will miss it.
closed 2023-03-26 by caveman99