meshtastic/firmware #2362: [Bug]: T-Beam with SX1268 not working LoRa on firmware >=2.1

PhilipLykov, 2023-03-17:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.1.2

### Description

After upgrade to version 2.1.0 the LoRa chip SX1268 is not working anymore. Even no configuration saves. After downgrade to 2.0.23 everything works well.  T-Beams on SX1278 are not affected.

caveman99, 2023-03-17:
There were a couple of radio fixes for the SX1262. We might need to validate if all were applied correctly to the SX1268 too.

GUVWAF, 2023-03-17:
Probably before it was actually using the SX1262 module of RadioLib and that was working fine. Now that RadioLib has a more strict chip detection method, this fails.

Can you try with PR #2363 when the artifacts is ready? This ensures that if it doesn’t find a SX1262, it will try the SX1268.

RJA74, 2023-03-17:
I came here to report the same issue. I have tried PR #2363 as above and it only partially resolves the issue for that I'm having. The radio now initializes and works but only when communicating with the same generation TBeam. 

I have 
2 x T-Beam with M8N & SX1262 and 
2 x T-Beam V1.1 with SX1276

Using firmware version v2.0.19.3209aea they all form a mesh
Using the above PR#2363 the SX1262's mesh with each other and the SX1276's mesh with each other but they don't communicate across the hardware versions.

I've just noticed as I was about to paste the logs in that the frequencies are different although the settings are identical:

### **SX1268**
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 2 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=12
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 4 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 3
INFO  | ??:??:?? 2 Radio frequency: _**433.174988**_
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=12
INFO  | ??:??:?? 2 SX126x init result 0
INFO  | ??:??:?? 2 Frequency set to 433.174988
INFO  | ??:??:?? 2 Bandwidth set to 250.000000
INFO  | ??:??:?? 2 Power output set to 12
DEBUG | ??:??:?? 2 Current limit set to 140.000000
DEBUG | ??:??:?? 2 Current limit set result 0
INFO  | ??:??:?? 2 Set Rx Power Saving Gain mode; result: 0
INFO  | ??:??:?? 2 SX1268 Radio init succeeded, using SX1268 radio

### **SX1278**
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 2 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=12
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 4 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 3
INFO  | ??:??:?? 2 Radio frequency: _**433.875000**_
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=12
DEBUG | ??:??:?? 2 Current limit set to 100.000000
DEBUG | ??:??:?? 2 Current limit set result 0
INFO  | ??:??:?? 2 RF95 init result 0
INFO  | ??:??:?? 2 Frequency set to 433.875000
INFO  | ??:??:?? 2 Bandwidth set to 250.000000
INFO  | ??:??:?? 2 Power output set to 12
INFO  | ??:??:?? 2 RF95 Radio init succeeded, using RF95 radio

Let me know if you want anymore of the logs.

GUVWAF, 2023-03-18:
@RJA74 Thanks for the logs. 
Indeed I could also see in the code it would always be overwritten. I think this module was thus not used for a long time and therefore this was not observed before. 
It needed to be overwritten, because if the region is still UNSET, it would try setting it to US frequency, which the chip can’t handle. Now it will only overwrite it when it is outside of the valid range, otherwise it will used the saved frequency.
Can you try with the new firmware of that PR? I don’t have an SX1268 myself, so I can’t test.

caveman99, 2023-03-18:
That code was committed by Lilygo when they introduced that chip to their lineup. I haven't seen someone actually use that chip until today ;-)

RJA74, 2023-03-18:
Thanks guys,

I confirm that cfa04d8 fixes the issue on the TBeam with SX1268 and is also working on the TBeam with SX1278.

Just want to say this is a great project and thanks to all who contribute.

PR #2363 (2023-03-18): SX126x: Try next Interface when chip not found
Potentially fixes #2362.
closed 2023-03-18 by thebentern