meshtastic/firmware #2364: T-Beam boot issue after being shut down from Android app

IZ1IVA, 2023-03-17:
Hi,

A T-Beam running firmware 2.1.2, after being shut down from Android app 2.1.1, will not boot up automatically when its USB port is plugged back into a power source. Pressing the POWER button does nothing, while pressing the USER button boots up the device.

This happens only when the device is shut down from the app, not from the POWER button.

Cheers,
Andrea

garthvh, 2023-03-17:
I was seeing similar issues with rak devices

rpsainio, 2023-03-17:
@IZ1IVA so, if you have USB-connected then you are able to run console and see what happens when you press USER-button. Most likely it will tell the wake-up reason. Might be a long Deep-Sleep allthough this is supposed to happen only on ESP32 platforms without AXP192

mverch67, 2023-03-17:
Same for TLora T3S3 !! Reboot from APP hangs and EVERY configuration change with reboot hangs. Need to press the reset button...

rpsainio, 2023-03-18:
Maybe so, still at least I would like to see some console output. This application is quite talkative :)

karamo, 2023-03-18:
T-Beam does shutdown with user-button, but suddenly reboots.

rpsainio, 2023-03-20:
The original problem seems to be that TBeam goes to DeepSleep instead switching itself off.
INFO  | ??:??:?? 272 From Radio onread
INFO  | ??:??:?? 277 Shutting down from admin command
INFO  | ??:??:?? 277 Shutting down
INFO  | ??:??:?? 277 Entering deep sleep for 18880 seconds
INFO  | ??:??:?? 277 Disable bluetooth
INFO  | ??:??:?? 277 GPS prepare sleep!
INFO  | ??:??:?? 277 GPS deep sleep!
INFO  | ??:??:?? 277 Saving /prefs/db.proto
INFO  | ??:??:?? 278 Saving /prefs/config.proto
INFO  | ??:??:?? 278 Saving /prefs/module.proto
INFO  | ??:??:?? 278 Saving /prefs/channels.proto
INFO  | ??:??:?? 279 Setting GPS power=0

@caveman99 is this now a feature of bug? See: https://github.com/meshtastic/firmware/pull/2352  Besides it does not sleep forever

karamo, 2023-03-20:
> T-Beam does shutdown with user-button, but suddenly reboots.

here the DEBUG-LOG:

```
DEBUG | 16:30:20 151424 [Button] Long press start!
...
INFO  | 16:30:25 151429 [Button] Turning off screen
INFO  | 16:30:25 151429 [Button] Shutting down
INFO  | 16:30:25 151429 [Button] Entering deep sleep for 18880 seconds
INFO  | 16:30:25 151429 [Button] GPS prepare sleep!
INFO  | 16:30:25 151429 [Button] GPS deep sleep!
DEBUG | 16:30:25 151429 [Button] sx126x entering sleep mode (FIXME, don't keep config)
INFO  | 16:30:25 151429 [Button] Saving /prefs/db.proto
INFO  | 16:30:25 151430 [Button] Saving /prefs/config.proto
INFO  | 16:30:26 151430 [Button] Saving /prefs/module.proto
INFO  | 16:30:26 151430 [Button] Saving /prefs/channels.proto
INFO  | 16:30:26 151430 [Button] Setting GPS power=0
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (71) esp_core_dump_flash: No core dump ðfÉÑ¥Ñ¥½¹found!
E (71) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   464][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
”ÃNFO  | ??:??:?? 0 

//\ E S H T /\ S T / C
INFO  | ??:??:?? 0 Booted, wake cause 3 (boot count 4), reset_reason=reset
```

rpsainio, 2023-03-22:
@garthvh  but as you can read it wakes up immediately if the button is kept pressed while ESP prepares for deepsleep. Very hard to implment that FR if at all - STILL a bug

rpsainio, 2023-04-10:
@caveman99 shall we open another issue for the fact that the deepsleep lasts only 18880 seconds, most likely on all ESP32 platoforms

garthvh, 2023-04-10:
> @caveman99 shall we open another issue for the fact that the deepsleep lasts only 18880 seconds, most likely on all ESP32 platoforms

There are no other logs with this number, only yours what value did you set for SDS?

rpsainio, 2023-04-10:
@garthvh you might notice that there are two logs in this issue from me and another one from Karamo. They both show the same value 18880 seconds when the deep sleep is entered. So mine was most likely 3600 seconds and you have to ask Karamo for that other log. I think this value might be a result of wrong conversion or simply a programming error leading to this funny value. So, we must build some debugging to see where the value gets set up.

karamo, 2023-04-11:
I am not a C/C++ specialist. But could this be the problem?

`sleep.cpp` #204:
```
void doDeepSleep(uint64_t msecToWake)
    LOG_INFO("Entering deep sleep for %lu seconds\n", msecToWake / 1000);
```

All values are defined as **UINT32**, but in `doDeepSleep` is **uint64_t** and format **%lu**.

caveman99, 2023-04-11:
we're setting the device to sleep for portMAX_DELAY milliseconds, which actually is a magic number if INCLUDE_vTaskSuspend is defined, meaning 'sleep forever'. This is documented in the FreeRTOS spec. The printed number is wrong, the code is doing the right thing.

caveman99, 2023-04-11:
@karamo #define portMAX_DELAY ( TickType_t ) 0xffffffffUL

karamo, 2023-04-11:
@caveman99 in` freertosinc.h` line 37:
`#define portMAX_DELAY UINT32_MAX`

rpsainio, 2023-04-11:
@caveman99  Let me test this as I recall that it came back after 18880 seconds.... and did not sleep for ever.

caveman99, 2023-04-11:
please wait a bit, fixing compilation errors ...

caveman99, 2023-04-11:
> @caveman99 in` freertosinc.h` line 37: `#define portMAX_DELAY UINT32_MAX`

this is not called if FreeRTOS is active. See a few line above that ... 
`// Include placeholder fake FreeRTOS defs`

The real def is in .platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\include\freertos\port\xtensa\include\freertos\portmacro.h

rpsainio, 2023-04-11:
anyway - test runs now

garthvh, 2023-04-11:
Makes sense that 18880 was a display error, it does not exist anywhere in all of GitHub other than this issue.

It also might make sense to set a SDS minimum value, anything less than 12 hours is essentially invalid and potentially dangerous, 24 hours is probably better. What would the real use case be for 2hours?  If solar ran flat overnight you are just firing up a flat battery again without any charging being done by solar.

The tbeam behavior has been updated to not shut down with the user button again, the screen brightness "feature" is totally broken and really should probably go away again. I have honestly never engaged it on purpose.

karamo, 2023-04-11:
@garthvh what do you mean? "18880 was a display error"
How can it be?

garthvh, 2023-04-11:
> 

Bad math, neither of you set a timer and  waited 5 and a half hours to see if it actually turned back on. I would reverse it, does anyone have any actual testing that indicates it was really going to wake from deep sleep after 5 hours?

rpsainio, 2023-04-11:
@garthvh I just tested this scenario and it did not turn on after 5 hours - so seems to be a problem with number formatting

garthvh, 2023-04-13:
Adding the screen brightness bug back is causing issues

tropho23, 2023-04-13:
New issue for this change: the primary problem is that it breaks user button initiated-shutdown, which is a critical feature for RAK-based devices.

In addition, this now doesn't even behave as intended; the  breaks screen function (via user button) of at least SH1106-based screens such as 1/3" OLEDs.  When long-pressing the user button, the brightness indicator bar briefly appears and the screen flickers and sometimes turns off, or dims (random and uncontrollable).  It takes a device reset to restore screen brightness and readability.

If this change resolved a T-Beam-specific issue, recommend keeping it *only* for T-Beam firmware and revert to the functionality & behavior from 2.1.7 for all other targets.

caveman99, 2023-04-13:
i am going to toss it again. this time for good :-) really more hassle than its worth.

tropho23, 2023-04-13:
Thanks!  As Garth noted, I doubt anyone even uses this capability and a reboot resets to screen default brightness anyways.  Dimming the screen would​ be a useful feature, if we were able to change a persistent brightness value that in the firmware/app settings somehow for low-light environments.  I don't want to distract from other development so this is not a feature request, just stating that would be a better way to adjust brightness, assuming it's even possible to do it.

Tony

caveman99, 2023-04-13:
next firmware will (again) remove screen brightness.

commit 29c13b5c (2023-04-11): resolve #2364
- fix wrong debug print
- change shutdown logic for t-beam if PMU is detected
- wait for 10 seconds instead of 5 for shutdown and resurrect screen brightness adjust for @karamo
closed 2023-04-13 by caveman99