meshtastic/firmware #2369: [Bug]:

MasterSPB, 2023-03-20:
### Category

Other

### Hardware

DIY

### Firmware Version

2.1.3

### Description

MCU freezes on init radio.

I have DYI hardware + E22 400M30S module (SX1268)
after flash firmware MCU work well, but when I setting region EU433 - MCU going to infinite loop with freeze and reboot by watchdog 

I changed couple modules -  i have always same trouble.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (479232/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (92 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (151 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 0  /static/index-0e816946.js.gz (399863 Bytes)
DEBUG | ??:??:?? 0  /static/index-9602d760.css.gz (17494 Bytes)
DEBUG | ??:??:?? 0  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   523][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   538][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=39, swver=2.1.4.36b00dba
DEBUG | ??:??:?? 0 Setting random seed 3460749087
DEBUG | ??:??:?? 0 Total heap: 196620
DEBUG | ??:??:?? 0 Free heap: 161872
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 90, FreeEntries 540, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 35
DEBUG | ??:??:?? 0 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 20
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 20
[   733][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 20
[   815][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 Number of Device Reboots: 35
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 2, using EU_433
DEBUG | ??:??:?? 0 region=2, NODENUM=0x5425bd44, dbsize=1
INFO  | ??:??:?? 0 Saving /prefs/db.proto
DEBUG | ??:??:?? 1 Read RTC time as 264
DEBUG | ??:??:?? 1 Using MSL altitude model
DEBUG | ??:??:?? 1 WANT GPS=1
WARN  | ??:??:?? 2 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 2 GxGSA NOT available
INFO  | ??:??:?? 2 External Notification Module Disabled
WARN  | ??:??:?? 2 SX126xInterface(cs=18, irq=33, rst=23, busy=32)
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 2 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=12
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 4 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 4
INFO  | ??:??:?? 2 Radio frequency: 433.875000
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=12
E (92248) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (92248) task_wdt:  - loopTask (CPU 1)
E (92248) task_wdt: Tasks currently running:
E (92248) task_wdt: CPU 0: IDLE
E (92248) task_wdt: CPU 1: loopTask
E (92248) task_wdt: Aborting.
```

GUVWAF, 2023-03-20:
Does this module and MCU work with firmware before 2.1.0? 
There is an open issue (#2208) with the same module, but it seems to fail at a different point now.

MasterSPB, 2023-03-21:
firmware 2.0.23 infinite loop with  reboot see this log:

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (479232/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (142 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 0  /static/index-71b9daa8.css.gz (17409 Bytes)
DEBUG | ??:??:?? 0  /static/index-d1cf5fb4.js.gz (399503 Bytes)
DEBUG | ??:??:?? 0  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (546 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   449][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=39, swver=2.0.21.83e6cea
DEBUG | ??:??:?? 0 Setting random seed 1041441772
DEBUG | ??:??:?? 0 Total heap: 189696
DEBUG | ??:??:?? 0 Free heap: 155092
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 2, FreeEntries 628, AllEntries 630, NameSpaces 1
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 8
DEBUG | ??:??:?? 0 OTA firmware version 0.2.0.1da2617
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 20
[   570][E][vfs_api.cpp:105] open(): /littlefs/prefs/config.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /prefs/config.proto preferences found
INFO  | ??:??:?? 0 Installing default LocalConfig
INFO  | ??:??:?? 0 Setting default channel and radio preferences!
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
[   603][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 20
[   683][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 Number of Device Reboots: 8
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
DEBUG | ??:??:?? 0 region=0, NODENUM=0xf2b7b53c, dbsize=1
INFO  | ??:??:?? 0 Saving /prefs/db.proto
DEBUG | ??:??:?? 0 Read RTC time as 0
DEBUG | ??:??:?? 0 Using MSL altitude model
DEBUG | ??:??:?? 0 WANT GPS=1
WARN  | ??:??:?? 1 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 1 GxGSA NOT available
INFO  | ??:??:?? 1 External Notification Module Disabled
INFO  | ??:??:?? 1 Starting meshradio init...
INFO  | ??:??:?? 1 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 19
INFO  | ??:??:?? 1 Radio frequency: 906.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=20
INFO  | ??:??:?? 2 RF95 init result -2
[... 55 lines cut ...]
DEBUG | ??:??:?? 42 Current limit set result -705
INFO  | ??:??:?? 47 Set Rx Power Saving Gain mode; result: 64831
WARN  | ??:??:?? 47 Failed to find LLCC68 radio
INFO  | ??:??:?? 47 Not using WIFI
ERROR | ??:??:?? 47 NOTE! Recording critical error 3 at�ɍ�main.cpp:513
INFO  | ??:??:?? 47 PowerFSM init, USB power=1
INFO  | ??:??:?? 47 Enter state: BOOT
[ 47549][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x402122ed  PS      : 0x00060630  A0      : 0x800e18d4  A1      : 0x3ffd8e60  
A2      : 0x3ffbd30c  A3      : 0x00000140  A4      : 0x00000001  A5      : 0x3f4245d5  
A6      : 0x3f42434f  A7      : 0x0000b9bd  A8      : 0x8013eef1  A9      : 0x3ffd8e50  
A10     : 0x00000000  A11     : 0x3ffd8e7f  A12     : 0x3ffcf094  A13     : 0x00060623  
A14     : 0x00060620  A15     : 0x00000001  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40093485  LEND    : 0x40093495  LCOUNT  : 0xffffffff  

Backtrace: 0x402122ea:0x3ffd8e60 0x400e18d1:0x3ffd8e90 0x400de6bc:0x3ffd8ec0 0x401312b9:0x3ffd8ee0

  #0  0x402122ea:0x3ffd8e60 in _svfwprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfwprintf.c:1284 (discriminator 85)
  #1  0x400e18d1:0x3ffd8e90 in SX126xInterface<SX1262>::reconfigure() at src/mesh/SX126xInterface.cpp:135
  #2  0x400de6bc:0x3ffd8ec0 in RotaryEncoderInterruptBase::runOnce() at src/input/RotaryEncoderInterruptBase.cpp:41
  #3  0x401312b9:0x3ffd8ee0 in spiEnableSSPins at /Users/masterspb/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:475
      (inlined by) spiEnableSSPins at /Users/masterspb/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:464

caveman99, 2023-03-21:
this kind of crash is often an incorrect pin assignment/setup. It's waiting for a state change that never happens, or double-uses I/O lines and tangles them up.

MasterSPB, 2023-03-21:
Hmm,
I checked MCU without connected modem on firmware 2.1.3 - and have same issue like in my first message 

....
INFO  | ??:??:?? 62 Slot time: 42 msec
INFO  | ??:??:?? 62 Set radio: final power level=22
E (92100) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (92100) task_wdt:  - loopTask (CPU 1)
E (92100) task_wdt: Tasks currently running:
E (92100) task_wdt: CPU 0: IDLE
E (92100) task_wdt: CPU 1: IDLE
E (92100) task_wdt: Aborting.

this mean current firmware have incorrect modem check

caveman99, 2023-03-25:
please try 2.1.4 - there was a change regarding EU_433

caveman99, 2023-04-02:
This was a bug in recent RadioLib - will be fixed in Firmware 2.1.7 or in current Master branch

closed 2023-04-02 by caveman99

SCWhite, 2023-05-05:
HI @caveman99 
I'm having same issue with firmware 2.1.11
using the "LLCC86" lora module still have some problem.

@ INFO  | ??:??:?? 1 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1  channels.proto (53 Bytes)
DEBUG | ??:??:?? 1  db.proto (146 Bytes)
DEBUG | ??:??:?? 1 Using analog input -1 for battery level
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
INFO  | ??:??:?? 1 No I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=40, swver=2.1.11.313860c8
DEBUG | ??:??:?? 1 Reset reason: 0x0
DEBUG | ??:??:?? 1 Setting random seed 2932533512
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
INFO  | ??:??:?? 1 Loaded saved devicestate version 20
INFO  | ??:??:?? 1 No /prefs/config.proto preferences found
INFO  | ??:??:?? 1 Installing default LocalConfig
INFO  | ??:??:?? 1 Setting default channel and radio preferences!
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 0, using UNSET
INFO  | ??:??:?? 1 No /prefs/module.proto preferences found
INFO  | ??:??:?? 1 Installing default ModuleConfig
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded saved channelFile version 20
INFO  | ??:??:?? 1 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 0, using UNSET
DEBUG | ??:??:?? 1 region=0, NODENUM=0x95dcf586, dbsize=1
DEBUG | ??:??:?? 1 Using MSL altitude model
DEBUG | ??:??:?? 1 WANT GPS=1
WARN  | ??:??:?? 2 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 2 GxGSA NOT available
INFO  | ??:??:?? 2 External Notification Module Disabled
WARN  | ??:??:?? 2 SX126xInterface(cs=6, irq=7, rst=27, busy=26)
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 2 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 20
INFO  | ??:??:?? 2 Radio frequency: 906.875000
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=22
INFO  | ??:??:?? 2 SX126x init result -2
WARN  | ??:??:?? 2 Failed to find SX1262 radio
WARN  | ??:??:?? 2 SX126xInterface(cs=6, irq=7, rst=27, busy=26)
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 2 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 20
INFO  | ??:??:?? 2 Radio frequency: 906.875000
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=22
INFO  | ??:??:?? 3 SX126x init result -707
INFO  | ??:??:?? 3 Frequency set to 906.875000
INFO  | ??:??:?? 3 Bandwidth set to 250.000000
INFO  | ??:??:?? 3 Power output set to 22
DEBUG | ??:??:?? 3 Current limit set to 140.000000
DEBUG | ??:??:?? 3 Current limit set result 0
INFO  | ??:??:?? 3 Set Rx Power Saving Gain mode; result: 0
WARN  | ??:??:?? 3 Failed to find LLCC68 radio
ERROR | ??:??:?? 3 NOTE! Recording critical error 3 at src\main.cpp:624
INFO  | ??:??:?? 3 PowerFSM