meshtastic/firmware #2372: Add rssi to rangetest.csv

gdt, 2023-03-21:
### Platform

ESP32

### Description

With firmware 2.0.22, I enabled rangetest and log to csv.   That worked as expected, but while I see snr in the file, I do not see rssi.  It seems obvious that rssi should be logged, and that this is likely a small change.

While manually recording rssi and snr, I observe that snr never gets particularly high.   Looking at rssi, it decreases with distance in an intuitive way (for someone used to UHF propagation).  But snr seems to have a more limited range.   As actual data:
```
rssi  snr  distance
-137 -65   845m
-112   9.5 374m
 -50  10     1m
```
This is off topic in a strict sense for this feature request, but given as an example to motivate that rssi is of interest.  At the very least, having both enables analyzing the relationship, which could help locate noise sources.

gdt, 2023-03-21:
I can see the point that RSSI is not always what people think it is.  But it should be logged because it is part of the puzzle of people trying to understand things.   Right now, it's simply not available to look at to understand.

More seriously, the snr values appear wrong.  With an rssi of -50, and the same -120 to -130 dBm noise, I should be seeing snrs around 70.  but I don't ; they seem to be clipped at 10.

Also, in my limited data of recording rssi and snr from the app's display, they are consistent at low levels, more or less showing noise in the range of -120 to -130 dBm.   So I do in fact see rssi levels below the noise floor.  (And I realize we need to talk about de-spread signal and bandwidth to really get this right.)

Finally, if rssi is deemed not meaningful, then I don't understand why the android app shows it in the first place.  Or for that matter, why the LoRa chip reports it.   I resolve this by believing that it is generally meaningful but at times tricky, and that it's better to just display and log the data from the chip, and let people do with it what they would like.   I see this as the same as the snr values, which are at times wrong, such as in my 1m distance case.

garthvh, 2023-03-21:
Pretty sure the LoRa SNR range is -20 to +10

There are lots of values output by various chips that require additional calculations to be valueable that don't get displayed to the user.

gdt, 2023-03-21:
I took another test run.   I was definitely seeing rssi below noise floor, -140ish, and rssi values were sensible and useful.  I don't understand refusing to log them, as people trying to analyze what's going on might want to look at them, and I certainly do.

garthvh, 2023-03-22:
https://github.com/meshtastic/firmware/issues/1215

mc-hamster, 2023-03-22:
> Pretty sure the LoRa SNR range is -20 to +10
> 
> There are lots of values output by various chips that require additional calculations to be valueable that don't get displayed to the user.

The scale of SNR is different depending on the different modem settings.  :)

mc-hamster, 2023-03-22:
> I took another test run. I was definitely seeing rssi below noise floor, -140ish, and rssi values were sensible and useful. I don't understand refusing to log them, as people trying to analyze what's going on might want to look at them, and I certainly do.

The number you're seeing doesn't represent the noise floor. 

The noise floor is a measured value and snr uses that as part of a formula to give proper readings.

While it is a small change, it will open up a can of confusion and reports of poor reception.

rssi doesn't really work well for our technology.

Closing.

closed 2023-03-22 by mc-hamster