meshtastic/firmware #2376: [Feature Request]: MQTT Does not support TLS Transport Encryption

ghostop14, 2023-03-23:
### Platform

ESP32

### Description

Configuring an MQTT server such as mosquitto to use TLS with a valid certificate for transport security fails to connect.  Looks like this isn't supported in the current implementation?

garthvh, 2023-03-24:
Nobody has managed to make it work. There is a comment in the firmware here 

https://github.com/meshtastic/firmware/blob/9d6e1ce8e5aa39ab5eb58619eca17cef00e946a9/src/mqtt/MQTT.h#L25 

@sachaw found this potential approach as well https://github.com/knolleary/pubsubclient/pull/851/files

ghostop14, 2023-03-24:
I have a working Mosquitto MQTT server here with a valid purchased cert and such, validated with mosquitto_pub/sub to be all around valid TLS with authentication enabled.  Running Mosquitto 2.0.11 on Ubuntu 20.04.  That side all works.

There's definitely the client certificate-based validation I see in the potential approach (lines 105-110 in that link).  But maybe even to start simpler, we can support the user/pass authentication to not have to worry about uploading client id certificates to the devices.  That should basically be just the extra TLS negotiation from the client-side.

So I think we probably need a couple of things:
1. Just support for enabling TLS from the client class
2. For the self-signed certs, we probably need something like mqtt.tls_insecure to not validate the CA on those self-signed certs.  Otherwise the handshake should fail for MiTM/spoofing security reasons (Default should be to require all-around valid/signed cert for security)
3. Maybe an mqtt.tls_enabled since there's already a payload encryption setting?
4. The default encrypted port is 8883.  So if tls is enabled, and no port is specified in the address, it should be changed to 8883 rather than the unencrypted 1883.

I'm happy to provide the encrypted MQTT server side and work with anyone who wants to tackle the firmware code.  It's a dev box, so I can switch between unencrypted, and TLS valid or self-signed as needed.  DM me on discord and we can coordinate connection details.

ghostop14, 2023-03-24:
This article has some example code:  https://arduino.stackexchange.com/questions/72684/how-to-connect-to-mqtt-broker-with-tls

May just need to add the secure client wrapper before the pubsubclient:
WiFiClientSecure espClient;     // <-- Change #1: Secure connection to MQTT Server 
PubSubClient client(espClient);

// Then something like this if mqtt.tls_insecure is set:
espClient.setInsecure();

caveman99, 2023-04-07:
will be in next firmware for ESP32

closed 2023-04-07 by caveman99