meshtastic/firmware #2379: [Bug]: LilyGo S3 T3S3 v1.1 fails to detect SX1280 radio and boot loops

porkcharsui, 2023-03-24:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.1.4.958d2cf

### Description

Attempted to boot with `firmware-tlora-t3s3-v1-2.1.4.958d2cf.bin` flashed to a new T3S3 V1.1. It appears the radio isn't being detected.

Related discussion: https://meshtastic.discourse.group/t/lilygo-t3s3-v1-1/7405/5

### Relevant log output

```Shell
$ meshtastic  --noproto
WARNING file:mesh_interface.py _sendToRadio line:511 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:389 Not sending packet because protocol use is disabled by noProto
WARNING file:mesh_interface.py _sendToRadio line:511 Not sending packet because protocol use is disabled by noProto
INFO  | ??:??:?? 3

//\ E S H T /\ S T / C

INFO  | ??:??:?? 3 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 3 Filesystem files (479232/1048576 Bytes):
DEBUG | ??:??:?? 3  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 3  /prefs/db.proto (142 Bytes)
DEBUG | ??:??:?? 3  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 3  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 3  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 3  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 3  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 3  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 3  /static/index-0e816946.js.gz (399863 Bytes)
DEBUG | ??:??:?? 3  /static/index-9602d760.css.gz (17494 Bytes)
DEBUG | ??:??:?? 3  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 3  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 3  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 3  /static/site.webmanifest.gz (196 Bytes)
DEBUG | ??:??:?? 3 Using analog input 1 for battery level
INFO  | ??:??:?? 3 Scanning for i2c devices...
DEBUG | ??:??:?? 3 Scanning for i2c devices on port 2
DEBUG | ??:??:?? 3 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 3 I2C device found at address 0x3c
INFO  | ??:??:?? 3 sh1106 display found
DEBUG | ??:??:?? 3 0x0 subtype probed in 1 tries
INFO  | ??:??:?? 3 1 I2C devices found
DEBUG | ??:??:?? 3 acc_info = 0
DEBUG | ??:??:?? 4 No SD_MMC card detected
INFO  | ??:??:?? 4 Meshtastic hwvendor=16, swver=2.1.4.958d2cf
DEBUG | ??:??:?? 4 Setting random seed 446949608
DEBUG | ??:??:?? 4 Total heap: 284856
DEBUG | ??:??:?? 4 Free heap: 242552
DEBUG | ??:??:?? 4 Total PSRAM: 2095103
DEBUG | ??:??:?? 4 Free PSRAM: 2095103
DEBUG | ??:??:?? 4 NVS: UsedEntries 2, FreeEntries 628, AllEntries 630, NameSpaces 1
DEBUG | ??:??:?? 4 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 4 Number of Device Reboots: 149
DEBUG | ??:??:?? 4 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 4 Initializing NodeDB
INFO  | ??:??:?? 4 Loading /prefs/db.proto
INFO  | ??:??:?? 4 Loaded saved devicestate version 20
INFO  | ??:??:?? 4 No /prefs/config.proto preferences found
INFO  | ??:??:?? 4 Installing default LocalConfig
INFO  | ??:??:?? 4 Setting default channel and radio preferences!
DEBUG | ??:??:?? 4 Expanding short PSK #1
INFO  | ??:??:?? 4 Wanted region 0, using UNSET
INFO  | ??:??:?? 4 No /prefs/module.proto preferences found
INFO  | ??:??:?? 4 Installing default ModuleConfig
INFO  | ??:??:?? 4 Loading /prefs/channels.proto
INFO  | ??:??:?? 4 Loaded saved channelFile version 20
INFO  | ??:??:?? 4 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 5 Number of Device Reboots: 149
DEBUG | ??:??:?? 5 Expanding short PSK #1
INFO  | ??:??:?? 5 Wanted region 0, using UNSET
DEBUG | ??:??:?? 5 region=0, NODENUM=0xfa5a1bb0, dbsize=1
INFO  | ??:??:?? 5 Saving /prefs/db.proto
DEBUG | ??:??:?? 5 Read RTC time as 4
DEBUG | ??:??:?? 5 Using MSL altitude model
DEBUG | ??:??:?? 5 WANT GPS=1
WARN  | ??:??:?? 6 Failed to find UBlox & MTK GNSS Module
DEBUG | ??:??:?? 6 GxGSA NOT available
INFO  | ??:??:?? 6 External Notification Module Disabled
INFO  | ??:??:?? 6 Turning on screen
DEBUG | ??:??:?? 6 Module wants a UI Frame
DEBUG | ??:??:?? 6 Module wants a UI Frame
INFO  | ??:??:?? 6 Starting meshradio init...
DEBUG | ??:??:?? 6 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 7 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 7 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 7 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 7 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 7 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 7 Radio channel_num: 20
INFO  | ??:??:?? 7 Radio frequency: 906.875000
INFO  | ??:??:?? 7 Slot time: 42 msec
INFO  | ??:??:?? 7 Set radio: final power level=20
INFO  | ??:??:?? 7 RF95 init result -2
INFO  | ??:??:?? 7 Frequency set to 906.875000
INFO  | ??:??:?? 7 Bandwidth set to 250.000000
INFO  | ??:??:?? 7 Power output set to 20
WARN  | ??:??:?? 7 Failed to find RF95 radio
WARN  | ??:??:?? 7 SX126xInterface(cs=7, irq=33, rst=8, busy=34)
INFO  | ??:??:?? 7 Starting meshradio init...
DEBUG | ??:??:?? 7 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 7 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 7 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 7 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 7 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 7 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 7 Radio channel_num: 20
INFO  | ??:??:?? 7 Radio frequency: 906.875000
INFO  | ??:??:?? 7 Slot time: 42 msec
INFO  | ??:??:?? 7 Set radio: final power level=22
INFO  | ??:??:?? 7 SX126x init result -2
WARN  | ??:??:?? 7 Failed to find SX1262 radio
INFO  | ??:??:?? 7 Starting meshradio init...
DEBUG | ??:??:?? 7 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 7 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 7 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 7 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 7 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 7 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 7 Radio channel_num: 20
INFO  | ??:??:?? 7 Radio frequency: 906.875000
INFO  | ??:??:?? 7 Slot time: 42 msec
INFO  | ??:??:?? 7 Set radio: final power level=13
INFO  | ??:??:?? 12 SX128x init result -705
INFO  | ??:??:?? 12 Frequency set to 906.875000
INFO  | ??:??:?? 12 Bandwidth set to 250.000000
INFO  | ??:??:?? 12 Power output set to 13
WARN  | ??:??:?? 12 Failed to find SX1280 radio
INFO  | ??:??:?? 12 Not using WIFI
ERROR | ??:??:?? 12 NOTE! Recording critical error 3 at src/main.cpp:630
INFO  | ??:??:?? 12 PowerFSM init, USB power=1
INFO  | ??:??:?? 12 Enter state: BOOT
WARNING file:stream_interface.py __reader line:179 Meshtastic serial port disconnected, disconnecting... read failed: [Errno 6] Device not configured
```

geeksville, 2023-03-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/lilygo-t3s3-v1-1/7405/6

GUVWAF, 2023-03-24:
Does this board have an actual SX1280 radio (2.4 GHz)? The T3S3 can have different radios.

porkcharsui, 2023-03-24:
Board used to produce that log was ordered here:
https://www.aliexpress.us/item/3256804440825086.html - specifically SX1280 2.4G With PA model

The silkscreen on the board reads T3S3 V1.1 2021205

[image]

caveman99, 2023-03-24:
the radio is initialized with the wrong frequency range. there's code in there that should switch to LORA_24 and reboot in that case, but a recent refactor to support the Sub-GHz chips might have invalidated that check.

caveman99, 2023-03-24:
```
INFO  | ??:??:?? 7 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 7 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 7 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 7 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 7 Radio channel_num: 20
INFO  | ??:??:?? 7 Radio frequency: 906.875000
INFO  | ??:??:?? 7 Slot time: 42 msec
INFO  | ??:??:?? 7 Set radio: final power level=13
INFO  | ??:??:?? 12 SX128x init result -705
```

GUVWAF, 2023-03-24:
@caveman99 Maybe we can do the same as for SX1268 radios? Overriding `getFreq()` to set it to 2.4GHz when there is no `saved_freq` yet, such that it initializes on the first attempt.

caveman99, 2023-03-24:
i chose the approach to switch region and reboot because BW setting is different in 2.4GHz too. The slots are not compatible and need to be remapped.

GUVWAF, 2023-03-24:
Ah, I see. Actually it is not rebooting now because the error returned by RadioLib is different. -705 is SPI command timeout. Invalid frequency should be -12.

caveman99, 2023-03-24:
yes, saw that as well. maybe we need to compare to rediolib_ok, since 1280 is tried to init last anyway.

mverch67, 2023-03-24:
It could also be a wrong DIO2 / BUSY assignment (e.g. 36 instead of 34). I saw random failures when testing with the SX1262 chipset in that case.

caveman99, 2023-03-24:
it's right according to the scematics (the 1280 uses a module called HPD-17PA) - what intrigues me is that the Interrupt TX example from Lilygo is working with radiolib 5.5.0 but not with 5.7.0 any more.

caveman99, 2023-03-24:
[image]

GUVWAF, 2023-03-31:
@caveman99 Seems related to https://github.com/jgromes/RadioLib/issues/716

caveman99, 2023-04-02:
> @caveman99 Seems related to [jgromes/RadioLib#716](https://github.com/jgromes/RadioLib/issues/716)

@GUVWAF indeed good catch - will be fixed in Firmware 2.1.7 or in current Master branch

closed 2023-04-02 by caveman99