meshtastic/firmware #2396: [Feature Request]: consider nodeinfo / nodedb entries stale at some point and clean it up

caveman99, 2023-03-30:
### Platform

NRF52, ESP32

### Description

At the moment nodedb will grow till the nodelimit is reached, and then the least heard entries will be purged. Introduce a new setting to delete nodes we haven't heard in -say- a week and make this threshold configurable. also introduce sensible defaults for the different device roles. Restore old (not very user friendly) behaviour by setting this new value to zero.

tavdog, 2023-09-26:
I think this would be nice to have.  I only recently discovered the "reset nodeDB" option in the app to clean up old or dead nodes.

thebentern, 2024-02-17:
This will be handled in the upcoming NodeDB TTL configuration

closed 2024-02-17 by thebentern