meshtastic/firmware #2401: [Feature Request]: Per-network support

tb0hdan, 2023-04-02:
### Platform

ESP32

### Description

Hi,

As one of the Ukrainian community members, I'd like to make a following proposal:

Set up a test environment at the developer side, with MQTT credentials from larger networks, with hardware, that is mostly
available within the regional Meshtastic network and use it to debug reboots, hangs and other issues. 

I am ready to pay for a device dedicated for this purpose. I'll send credentials using secure means when necessary.

Thanks.

garthvh, 2023-04-02:
What exactly do you need?

tb0hdan, 2023-04-03:
In our specific case, improved reliability is required. I.e. fewer reboots for MQTT nodes. Right now it ranges from 1-2 per hour on TBEAM devices, which we're using as routers.

tb0hdan, 2023-04-04:
Plus, select developers will definitely benefit from having real traffic on their devices

tb0hdan, 2023-04-27:
Any updates on this? Our network seems to be the biggest so far with the highest amount of traffic and consecutive problems.

mc-hamster, 2023-04-27:
Not sure of the request.

Is this request for dedicated mqtt servers pre region?

tb0hdan, 2023-04-27:
Our network is already larger than `mqtt.meshtastic.org` is. Both overall and online nodes. I need someone from core developers to reach out, I'll provide MQTT credentials for testing new firmware and debugging issues on big networks. That's what I'm asking for.

ut3uku, 2023-04-27:
Traffic is a major problem in Ukraine's network. Messages are being delayed or dropping. We cannot rely on it for emergency communication((

garthvh, 2023-04-28:
Step one is to disable JSON over MQTT, it is not for production.

Step two is to turn down your hop limit if it has been pushed to 7.

The next step is to post a config from one of your connected devices so we can take a look.

tb0hdan, 2023-04-29:
Hey guys, MQTT is now combined, with 73 nodes online. You should get some nice traffic.

tb0hdan, 2023-04-29:
[image]

tb0hdan, 2023-04-29:
2.1.7 / T-Beam

- 67/80 nodes
-  node_db full erasing... message
- no reboots though

tb0hdan, 2023-04-29:
[image]

Only one message was sent. Duplicates.

tb0hdan, 2023-04-29:
Also, most attempts to run `meshtastic --nodes` fail with "timeout" error, but when they are successful (rarely), device stops responding to ping:

```
64 bytes from 192.168.3.6: icmp_seq=267 ttl=255 time=3.94 ms
64 bytes from 192.168.3.6: icmp_seq=268 ttl=255 time=2.29 ms

64 bytes from 192.168.3.6: icmp_seq=278 ttl=255 time=3.16 ms
64 bytes from 192.168.3.6: icmp_seq=279 ttl=255 time=2.98 ms
```

tb0hdan, 2023-04-29:
Okay, got it. Every time any kind of serial command succeeds, device reboots itself

tb0hdan, 2023-04-29:
[image]

tb0hdan, 2023-04-29:
```

tb0hdan, 2023-04-29:
```
│  75 │ Gremlen_AFR                  │ 5c2c  │ !a3fb5c2c │ N/A        │ N/A         │ N/A        │ N/A       │ N/A       │ N/A                 │ N/A            │
```

NodeDB doesn't reach more than 75 nodes ever

tb0hdan, 2023-04-29:
2.1.7 reboots every 44 seconds with current node count

garthvh, 2023-04-29:
With a large network --info and --nodes are unlikely to work reliably in the python CLI, both of these methods aggregate a bunch of different packets together and is slow. You might try the C# Cli as it probably performs better.

garthvh, 2023-04-29:
Unless you are using short fast  I would not anticipate getting to 80 nodes.

Nodes like this one are going to create a lot of traffic, 7 hops every minute with many nodes really multiplies quickly. https://mesh.in.ua/grafana/d/Q2DodHa4k/mesh-node?orgId=1&var-nodeId=917327552&refresh=10s

garthvh, 2023-04-29:
Per @caveman99 The problem with high hop limits is the packet is cascading and multiplying. If you have every router in reach in an ideal network one packet on Hop Limit 7 causes 126 retransmissions.  Unnecessarily high hop limits on some nodes combined with excess telemetry and position packets is likely most of your problem (wifi on a tbeam has issues as well)

You should also make sure that router router client or repeater are only used on stationary nodes with an advantage of some kind (power, height, antenna etc) if everything is a router then there is no advantage to using that mode at all.

garthvh, 2023-12-16:
Network support is not something the project will be providing.  @jp-bennett may have some interest.

closed 2023-12-16 by garthvh