meshtastic/firmware #2415: [Bug]: RAK Buzzer does not alert on bell character

garthvh, 2023-04-06:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.1.7

### Description

With External notifications enabled and the buzzer pin set to 21 when alert on bell character and use pwm are enabled there is no alert on the buzzer

buzztiaan, 2023-04-06:
I dont think the external notifications work at all. 
I cant find any method that lets the buzzer beep on receiving messages.

garthvh, 2023-04-06:
For the rak buzzer you should set buzzer pin to 21 on a 19007 and check the use pwm toggle at the top, those GPIO settings are all meshenger.

garthvh, 2023-04-06:
@caveman99 any idea when we turned of the buzzer for messages?  I am definitely sending the bell character right as ben's CLI beeps ever time it renders one.

buzztiaan, 2023-04-06:
Preferences: { "device": { "buzzerGpio": 14 }   ,    "externalNotification": { "alertMessage": true, "usePwm": true }

that's all thats needed right? (i just threw it on a random pin, as said it plays a tune on booting just fine)

caveman99, 2023-04-11:
@garth external_notification.enabled obviously needs to be on, then it parses incoming payloads for 0x07

we define three different notification behaviours 'on Bell' (external_notification.alert_bell, external_notification.alert_bell_vibra and external_notification.alert_bell_buzzer) but only external_notification.alert_bell_buzzer will decide whether the buzzer is a PWM type.

The other two will just set the gpio static to support motor and LED output.

caveman99, 2023-04-11:
see https://github.com/meshtastic/firmware/blob/320bf5768753e5423dd607e0792a6e99a3d31c46/src/modules/ExternalNotificationModule.cpp#L250-L265 and https://github.com/meshtastic/firmware/blob/320bf5768753e5423dd607e0792a6e99a3d31c46/src/modules/ExternalNotificationModule.cpp#L289-L302

buzztiaan, 2023-04-11:
could if (moduleConfig.external_notification.alert_message)  please get actual pwm notification?

garthvh, 2023-04-12:
@caveman99 so having bell and PWM on with the device config GPIO should be all I need right? 

I can only get the PWM buzzer to do startup and shutdown, I get nothing for messages.

buzztiaan, 2023-04-16:
finally got it!
i noticed a blob of parameters on https://github.com/meshtastic/firmware/blob/320bf5768753e5423dd607e0792a6e99a3d31c46/src/modules/ExternalNotificationModule.cpp#L148

so;

> meshtastic --set external_notification.output_ms 1000
> meshtastic --set external_notification.nag_timeout 300
> meshtastic --set external_notification.enabled True
> meshtastic --set external_notification.alert_message_buzzer True
> meshtastic --set external_notification.use_pwm True

> meshtastic --set-ringtone Looney:d=4,o=5,b=140:32p,c6,8f6,8e6,8d6,8c6,a.,8c6,8f6,8e6,8d6,8d#6,e.6,8e6,8e6,8c6,8d6,8c6,8e6,8c6,8d6,8a,8c6,8g,8a#,8a,8f

and i'm getting looneytunes on msgs i have received! yay! 

nag_timeout seems in seconds, its how long the ringtone will bleep and bloop, you can press the button to stop it beeping

kingoffoxes, 2023-11-15:
i found i would only get clicks at the start and end of the expected buzzing period - as if it was just set high and then low. 
This was using the iOS app to apply the settings. 

When connected via python everything worked as it should and played the tune after i set this command. 

    meshtastic --set device.buzzer_gpio 13

(i had also specified pin 13 in the iOS app)

garthvh, 2023-12-23:
Verified as working

closed 2023-12-23 by garthvh