meshtastic/firmware #2416: [Bug]: sporadic crashes when sending text messages via phone

mverch67, 2023-04-06:
### Category

Other

### Hardware

T-Lora v1.3

### Firmware Version

2.1.7

### Description

When sending text messages via phone the device crashes and reboots. This happens quite often!

I suspect that the channel crypto library is not reentrant and cannot handle two send or receive requests at the same time (getKey/setKey/setCrypto), which leads to possible memory corruption and eventually a crash (w/o stacktrace on esp32-S3).

The channel crypto methods should be guarded with a concurrency lock.

garthvh, 2023-04-07:
May be related to https://github.com/meshtastic/firmware/issues/2078

garthvh, 2023-04-07:
I have seen it with S3 boards

mverch67, 2023-04-07:
Yeah, it's the same issue. S3 boards seem to be more sensitive during memory corruption or other invalid operations. 
E.g. 1 << 40 (which is more than 32 bit) will crash on S3 but not on plain esp32 microcontrollers.

But the main cause in this issue and #2078 is always the same: The device is sending or receiving a msg (e.g. telemetry or nodeinfo) when it gets a packet from phone and tries to encrypt it.

You see the interface of CryptoEngine is not made for concurrency: first you have to set the key, then call the encrypt/decrypt method. If another taks switches in between... booom!

void CryptoEngine::setKey(const CryptoKey &k)
void CryptoEngine::encrypt(uint32_t fromNode, uint64_t packetId, size_t numBytes, uint8_t *bytes)
void CryptoEngine::decrypt(uint32_t fromNode, uint64_t packetId, size_t numBytes, uint8_t *bytes)

You can make this crash more deterministic by inserting a longer delay() inside Channels::setCrypto() between getKey() and setKey().

Niximkk, 2023-04-07:
True, same here... its really sad since i dont have a notebook to change some configs. Im camping rn :)

mverch67, 2023-04-08:
The commit #2420 helps preventing that encode/decode run in parallel threads at the same time.

However, the must be still an issue within the encrypt code where still a crash happens when rapidly sending messages from phone:

`DEBUG | 08:21:46 207 PACKET FROM PHONE (id=0x77bffcd9 fr=0x00 to=0x30, WantAck=1, HopLim=0 Ch=0x0 Portnum=1)
DEBUG | 08:21:46 207 localSend to channel 0
DEBUG | 08:21:46 207 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=23, time 509 ms
DEBUG | 08:21:46 207 Setting next retransmission in 7030 msecs:  (id=0x77bffcd9 fr=0x00 to=0x30, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1680942106)
DEBUG | 08:21:46 207 Add packet record (id=0x77bffcd9 fr=0x00 to=0x30, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1680942106)
DEBUG | 08:21:46 207 ------------------------ encode critical path start ------------------------
DEBUG | 08:21:46 207 Original length - 3 
DEBUG | 08:21:46 207 Compressed length - 4 
DEBUG | 08:21:46 207 Original message - Msg 
DEBUG | 08:21:46 207 Not using compressing message.
DEBUG | 08:21:46 207 Using AES256 key!
DEBUG | 08:21:46 207 ESP32 crypt fr=fa5a06bc, num=77bffcd9, numBytes=7!
DEBUG | 08:21:46 207 ------------------------ encode critical path end ---------------------------
DEBUG | 08:21:46 207 enqueuing for send (id=0x77bffcd9 fr=0xbc to=0x30, WantAck=1, HopLim=3 Ch=0x63 encrypted rxtime=1680942106)
DEBUG | 08:21:46 207 txGood=7,rxGood=9,rxBad=0
INFO  | 08:21:46 207 Telling client we have new packets 12
INFO  | 08:21:46 207 BLE notify fromNum
INFO  | 08:21:47 207 [RadioIf] From Radio onread
INFO  | 08:21:47 207 [RadioIf] getFromRadio=STATE_SEND_PACKETS
DEBUG | 08:21:47 207 [RadioIf] encoding toPhone packet to phone variant=11, 12 bytes
DEBUG | 08:21:47 207 [RadioIf] Starting low level send (id=0x77bffcd9 fr=0xbc to=0x30, WantAck=1, HopLim=3 Ch=0x63 encrypted rxtime=1680942106 priority=70)
DEBUG | 08:21:47 208 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=23, time 509 ms
DEBUG | 08:21:47 208 [RadioIf] AirTime - Packet transmitted : 509ms
INFO  | 08:21:47 208 From Radio onread
DEBUG | 08:21:47 208 [RadioIf] Completed sending (id=0x77bffcd9 fr=0xbc to=0x30, WantAck=1, HopLim=3 Ch=0x63 encrypted rxtime=1680942106 priority=70)
DEBUG | 08:21:48 209 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | 08:21:48 209 [RadioIf] Lora RX (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x63 encrypted rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:48 209 [RadioIf] AirTime - Packet received : 559ms
DEBUG | 08:21:48 209 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | 08:21:48 209 [Router] Add packet record (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x63 encrypted rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:48 209 [Router] ------------------------ decode critical path start ------------------------
DEBUG | 08:21:48 209 [Router] Using channel 0 (hash 0x63)
DEBUG | 08:21:49 210 [Router] Using AES256 key!
DEBUG | 08:21:49 210 [Router] ESP32 crypt fr=6d088430, num=1315e821, numBytes=11!
DEBUG | 08:21:49 210 [Router] decoded message (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 [Router] ------------------------ decode critical path end ---------------------------
DEBUG | 08:21:49 210 [Router] handleReceived(REMOTE) (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 [Router] Module 'routing' wantsPacket=1
INFO  | 08:21:49 210 [Router] Received routing from=0x6d088430, id=0x1315e821, portnum=5, payloadlen=2
DEBUG | 08:21:49 210 [Router] Routing sniffing (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 [Router] Received an ack for 0x77bffcd9, stopping retransmissions
DEBUG | 08:21:49 210 [Router] cancelSending id=0x77bffcd9, removed=0
DEBUG | 08:21:49 210 [Router] Delivering rx packet (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 [Router] Update DB node 0x6d088430, rx_time=1680942108, channel=0
DEBUG | 08:21:49 210 [Router] Forwarding to phone (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 [Router] ------------------------ decode critical path start ------------------------
DEBUG | 08:21:49 210 [Router] ------------------------ decode critical path end (abort)--------------------
DEBUG | 08:21:49 210 [Router] Module 'routing' considered
INFO  | 08:21:49 210 Telling client we have new packets 13
INFO  | 08:21:49 210 BLE notify fromNum
INFO  | 08:21:49 210 From Radio onread
INFO  | 08:21:49 210 getFromRadio=STATE_SEND_PACKETS
DEBUG | 08:21:49 210 phone downloaded packet (id=0x1315e821 fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffcd9 rxtime=1680942108 rxSNR=5.75 rxRSSI=-51)
DEBUG | 08:21:49 210 encoding toPhone packet to phone variant=2, 53 bytes
INFO  | 08:21:49 210 From Radio onread
DEBUG | 08:22:05 226 [Power] Battery: usbPower=1, isCharging=1, batMv=4453, batPct=0
INFO  | 08:22:10 231 [DeviceTelemetryModule] (Sending): air_util_tx=0.186194, channel_utilization=1.780000, battery_level=101, voltage=4.453000
[... cut ...]
DEBUG | 08:22:11 232 [Router] decoded message (id=0x53fbecce fr=0x30 to=0xbc, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1680942131 rxSNR=5.75 rxRSSI=-53)
DEBUG | 08:22:11 232 [Router] ------------------------ decode critical path end ---------------------------
DEBUG | 08:22:11 232 [Router] ------------------------ encode critical path start ------------------------
DEBUG | 08:22:11 232 [Router] Original length - 1 
DEBUG | 08:22:11 232 [Router] Compressed length - 2 
DEBUG | 08:22:11 232 [Router] Original message - F 
DEBUG | 08:22:11 232 [Router] Not using compressing message.

[10:22:18.124] Disconnected`

mverch67, 2023-04-08:
There is a call to setCryptoKey outside of the guarded lock which cause the crash:

```
DEBUG | ??:??:?? 80 [Router] ------------------------ decode critical path start ------------------------
DEBUG | ??:??:?? 80 [Router] ------------------------ decode critical path end (abort)--------------------
DEBUG | ??:??:?? 80 [Router] Module 'routing' considered
INFO  | ??:??:?? 80 Telling client we have new packets 35
INFO  | ??:??:?? 80 BLE notify fromNum
WARN  | ??:??:?? 81 [RadioIf] Can not send yet, busyRx
INFO  | ??:??:?? 81 From Radio onread
INFO  | ??:??:?? 81 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 81 phone downloaded packet (id=0x1315e88a fr=0x30 to=0xbc, WantAck=0, HopLim=3 Ch=0x0 Portnum=5 requestId=77bffce9 rxSNR=5.75 rxRSSI=-53)
DEBUG | ??:??:?? 81 [Screen] encoding toPhone packet to phone variant=2, 48 bytes
WARN  | ??:??:?? 81 [RadioIf] Can not send yet, busyRx
DEBUG | ??:??:?? 81 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | ??:??:?? 81 [RadioIf] Lora RX (id=0x1315e88a fr=0x30 to=0xbc, WantAck=0, HopLim=2 Ch=0x63 encrypted rxSNR=6.5 rxRSSI=-47)
DEBUG | ??:??:?? 81 [RadioIf] AirTime - Packet received : 559ms
DEBUG | ??:??:?? 81 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | ??:??:?? 81 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | ??:??:?? 81 [Router] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=27, time 559 ms
DEBUG | ??:??:?? 81 [Router] Found existing packet record for fr=0x6d088430,to=0xfa5a06bc,id=0x1315e88a
DEBUG | ??:??:?? 81 [Router] Found existing packet record for fr=0x6d088430,to=0xfa5a06bc,id=0x1315e88a
DEBUG | ??:??:?? 81 [Router] Add packet record (id=0x1315e88a fr=0x30 to=0xbc, WantAck=0, HopLim=2 Ch=0x63 encrypted rxSNR=6.5 rxRSSI=-47)
DEBUG | ??:??:?? 81 [Router] Ignoring incoming msg, because we've already seen it (id=0x1315e88a fr=0x30 to=0xbc, WantAck=0, HopLim=2 Ch=0x63 encrypted rxSNR=6.5 rxRSSI=-47)
DEBUG | ??:??:?? 81 [Router] Incoming message was filtered 0x6d088430
DEBUG | ??:??:?? 81 [RadioIf] Starting low level send (id=0x77bffcea fr=0xbc to=0x30, WantAck=1, HopLim=3 Ch=0x63 encrypted priority=70)
DEBUG | ??:??:?? 81 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=21, time 485 ms
DEBUG | ??:??:?? 81 [RadioIf] AirTime - Packet transmitted : 485ms
DEBUG | ??:??:?? 81 *** setCrypto set new key.length=32 ***
D
[11:09:51.230] Disconnected
```

mverch67, 2023-04-08:
It's not possible to reproduce the crash with TBeam 1.0 SX1262. It's stable.

Alpha-Ninja-4518, 2023-04-08:
I also got this issue a lot on T3S3 v1.1 on the latest beta.

Alpha-Ninja-4518, 2023-04-08:
Maybe do CPU software encode for sending messages and leave hardware for bluetooth? It can't be that much work encrypting messages as it's only AES and less than 256 bytes.

mverch67, 2023-04-08:
> I also got this issue a lot on T3S3 v1.1 on the latest beta.

did you try this firmware with the locks?
https://github.com/meshtastic/artifacts/blob/device/pr/pr2420-firmware-2.1.8.50f8915.zip?raw=true

Alpha-Ninja-4518, 2023-04-08:
> > I also got this issue a lot on T3S3 v1.1 on the latest beta.
> 
> did you try this firmware with the locks? https://github.com/meshtastic/artifacts/blob/device/pr/pr2420-firmware-2.1.8.50f8915.zip?raw=true

Not yet but above it suggests the problem still happens although less

caveman99, 2023-04-09:
setCrypto() boils down to the perhapsEncode and perhapsDecode functions which are fully locked by the new LockGuard. decryptForHash and setActiveByIndex are also only called from these 2 functions. I have no idea where else the calls may come from. Or if something else is crashing here.

thebentern, 2023-04-10:
> setCrypto() boils down to the perhapsEncode and perhapsDecode functions which are fully locked by the new LockGuard. decryptForHash and setActiveByIndex are also only called from these 2 functions. I have no idea where else the calls may come from. Or if something else is crashing here.

I'm honestly in favor of going ahead and integrating that lock PR even it doesn't solve _all_ of the issue across platforms. We can keep this issue open though to continue our investigationg

caveman99, 2023-04-11:
I pushed the initial lock code. Let's keep our collective eyes peeled...

caveman99, 2023-04-13:
next firmware will (again) remove screen brightness.

caveman99, 2023-04-13:
dang. wrong issue.

PR #2618 (2023-07-17): Fix: increase BT NIMBLE task stack size by 1k
Stacksize for task nimble_host is too small and leads to crashes when sending text messages from phone:

```
DEBUG | 21:27:08 12 [Screen] Add packet record (id=0xf0f7443c fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1689542828)
DEBUG | 21:27:08 12 Original length - 2 
DEBUG | 21:27:08 12 [GPS] Compressed length - 2 
DEBUG | 21:27:08 12 [GPS] Original message - Hi 
DEBUG | 21:27:08 12 [GPS] Not using compressing message.
DEBUG | 21:27:08 12 [GPS] Using AES256 key!

***ERROR*** A stack overflow in task nimble_host has been detected.
` [... cut]
closed 2023-07-17 by thebentern