meshtastic/firmware #2434: DeviceConfig 'Button Pin Override' not working

costonisp, 2023-04-17:
I tried to set pin 13 as a pin button for my T-beam (hardware T22 v1.0)  but it does not reconfigure this pin.

costonisp, 2023-04-17:
I see, if it's not in the firmware ..yet.., that must also be the reason that PinButton reconfiguration neither works in the python CLI

caveman99, 2023-04-18:
The Settings for Button, Buzzer and the RX/TX GPS GPIO lines were introduced at the same time. The button turned out to be a bit of a hassle to implement, so it got left on the side. I'll see if i can wire it up for the 2.1.10 firmware release.

caveman99, 2023-04-18:
@costonisp can you try with this firmware: https://github.com/meshtastic/artifacts/blob/device/pr/pr2439-firmware-2.1.10.9eb7a4c.zip?raw=true

costonisp, 2023-04-19:
I burned this 2.1.10 firmware and could reconfigure the Button Pin succesfully both with the web interface and with the python CLI.
tnx alot for your dedication.

I will close the issue as solved

commit 9e95df8f (2023-04-21): implement dynamic userbutton overwrite. fix #2434
commit e0bb95ca (2023-04-21): implement dynamic userbutton overwrite. fix #2434
closed 2023-04-19 by costonisp