meshtastic/firmware #2440: [Bug]: Crash when sending commands via serial

micheljung, 2023-04-18:
### Category

Serial

### Hardware

Heltec v2

### Firmware Version

2.1.9

### Description

1. Connect via USB
2. Use the CLI client 2.1.6
3. Send any command, e. g. `meshtastic.exe --info`
4. Watch the device crash/restart

### Relevant log output

```Shell
Connected to radio
```

caveman99, 2023-04-18:
It does not crash, but restarts. That is a shortcoming of the python serial libs and the hardware design of the heltec boards. The m5stack also does that. There's no easy solution and if there was one it would have to go into the python client. Try using the c-Sharp CLI instead, that should work better.

closed 2023-04-18 by caveman99

micheljung, 2023-04-18:
Thanks, just found this information:

[image]
https://meshtastic.org/docs/settings/moduleconfig/mqtt