meshtastic/firmware #2441: [Bug]: 2.1.7 tries to go to sleep and then crashes  crashes with WiFi enabled

tb0hdan, 2023-04-18:
### Category

WiFi

### Hardware

DIY

### Firmware Version

2.1.7

### Description

 2.1.7 crashes with WiFi enabled. Log shows button pressed but none buttons are soldered

### Relevant log output

```Shell
DEBUG | 17:20:22 57 [Button] Long press stop!
INFO  | 17:20:25 60 [Button] Shutting down
INFO  | 17:20:25 60 [Button] Entering deep sleep for 18880 seconds
DEBUG | 17:20:25 60 [Button] radio wait to sleep, txEmpty=0
... [ same message ] ...
DEBUG | 17:20:55 90 [Button] radio wait to sleep, txEmpty=0
ERROR | 17:20:55 90 [Button] NOTE! Recording critical error 2 at src/sleep.cpp:167

assert failed: void waitEnterSleep() sleep.cpp:168 (0)

Backtrace: 0x40084445:0x3ffd82c0 0x4009695d:0x3ffd82e0 0x4009ce35:0x3ffd8300 0x400fad19:0x3ffd8430 0x400fae5a:0x3ff
d8450 0x400d433d:0x3ffd8470 0x400df01d:0x3ffd8490 0x400fd78e:0x3ffd84b0 0x400fd7c9:0x3ffd84d0 0x400def54:0x3ffd84f0
 0x400d5d52:0x3ffd8510 0x400fe2c5:0x3ffd8530 0x400df5c6:0x3ffd8560 0x40134cb1:0x3ffd8580

ELF file SHA256: 6fd1560c427b8461

E (1081) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (813) esp_core_dump_flash: No core dump^@^@<85><C9>ѥѥ<BD><B9><81>found!
E (813) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
<94><C3>^@^B@^AINFO  | ??:??:?? 0

... BOOT Messages follow ...
```
```

garthvh, 2023-04-19:
0x40084445: print_entry at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/arch/xtensa/debug_helpers.c:52
0x40084445: esp_backtrace_print_from_frame at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/arch/xtensa/debug_helpers.c:77
0x4009695d: rtc_sleep_low_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/port/esp32/rtc_sleep.c:237
0x4009ce35: insert_free_block at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf.c:239
0x4009ce35: block_insert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf.c:263
0x4009ce35: tlsf_add_pool at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_tlsf.c:774
0x400fad19: JSON::ExtractString(char const**, std::__cxx11::basic_string, std::allocator >&) at /home/runner/work/firmware/firmware/src/mqtt/JSON.cpp:161
0x400fae5a: JSON::ParseDecimal(char const**) at /home/runner/work/firmware/firmware/src/mqtt/JSON.cpp:242
0x400d433d: XPowersAXP192::getDC2Voltage() at /home/runner/work/firmware/firmware/.pio/libdeps/tbeam/XPowersLib/src/XPowersAXP192.tpp:815
0x400d433d: XPowersAXP192::getPowerChannelVoltage(unsigned char) at /home/runner/work/firmware/firmware/.pio/libdeps/tbeam/XPowersLib/src/XPowersAXP192.tpp:1862
0x400df01d: drawCompassNorth at /home/runner/work/firmware/firmware/src/graphics/Screen.cpp:737
0x400df01d: drawNodeInfo at /home/runner/work/firmware/firmware/src/graphics/Screen.cpp:825
0x400fd78e: std::__cxx11::basic_string, std::allocator >::~basic_string() at /home/runner/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:656
0x400fd78e: MQTT::downstreamPacketToJson[abi:cxx11](_meshtastic_MeshPacket*) at /home/runner/work/firmware/firmware/src/mqtt/MQTT.cpp:516
0x400fd7c9: std::__cxx11::basic_string, std::allocator >::~basic_string() at /home/runner/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:657
0x400fd7c9: MQTT::downstreamPacketToJson[abi:cxx11](_meshtastic_MeshPacket*) at /home/runner/work/firmware/firmware/src/mqtt/MQTT.cpp:539
0x400def54: drawNodeInfo at /home/runner/work/firmware/firmware/src/graphics/Screen.cpp:813
0x400d5d52: XPowersAXP2101::disablePowerOutput(unsigned char) at /home/runner/work/firmware/firmware/.pio/libdeps/tbeam/XPowersLib/src/XPowersAXP2101.tpp:2686
0x400fe2c5: MQTT::onPublish(char*, unsigned char*, unsigned int) at /home/runner/work/firmware/firmware/src/mqtt/MQTT.cpp:65
0x400df5c6: graphics::Screen::handleSetOn(bool) at /home/runner/work/firmware/firmware/src/graphics/Screen.cpp:916
0x400df5c6: graphics::Screen::handleSetOn(bool) at /home/runner/work/firmware/firmware/src/graphics/Screen.cpp:907
0x40134cb1: std::function::~function() at /home/runner/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:370
0x40134cb1: WiFiEventCbList::~WiFiEventCbList() at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:742
0x40134cb1: WiFiGenericClass::onEvent(std::function, arduino_event_id_t) at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:839

Would recommend not using the unstable JSON MQTT mode

caveman99, 2023-04-19:
Can we get more info on the hardware? You say it's DIY. The log says you shout down the board with a button long press, so some (maybe not soldered) button pin is floating and picking up static... 
The Shutdown with longpress is indeed new in this firmware.

mverch67, 2023-04-20:
In case the device role was configured as ROUTER then the button press could possibly be a negative side effect of code changes in #2403 which was corrected by #2412 in firmware 2.1.8.

thebentern, 2023-05-26:
This issue seems to be resolved. Closing. Re-open if this is not the case

closed 2023-05-26 by thebentern