meshtastic/firmware #2443: [Bug]: Again, the story with the blue label | Observations

SR7673, 2023-04-19:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.1.9

### Description

Today i tried 2.1.9 on a RAK4631. It works (as far as observed now).
Then i wanted to flash the same firmware to another RAK4631 but that didn't work! 
Why? 
I then noticed that the working one has that blue sticker/label, just like described here: https://meshtastic.discourse.group/t/rak4631-rak19007-rf-switch-problem/7084/24
The ones without the sticker only work when using the rak4631_eink FW (tested with FW 2.1.5) here.

Furthermore i observed that some of my RAK4631 nodes, all configured as ROUTER_CLIENT, crashed after a while when using 2.1.5_rak4631_eink, while others did not. I now suspect that the crashing ones have the blue sticker.
That's no 100% analysis but maybe something worth to put an eye on for the experts...

SR7673, 2023-04-19:
I confirm, 2.1.9_eink is flashable on the module without the sticker.

garthvh, 2023-04-19:
The blue label indicates that you bought the arduino version

closed 2023-04-19 by garthvh

SR7673, 2023-04-19:
And the arduino version is the right one, right?
So is that the RAK4631 vs. RAK4631-R story? Both of my orders have been to RAK4631 and both of them are labeled with RAK4630...

garthvh, 2023-04-19:
It does indicate an older core, if you used it with 1.3 it is probably worn out.