meshtastic/firmware #2445: [Feature Request]: Support pinecone traffic/messages

alistair23, 2023-04-24:
### Platform

NRF52, ESP32

### Description

> Pinecone is an experimental overlay routing protocol suite which is the foundation of the current P2P Matrix demos. It is designed to provide end-to-end encrypted connectivity between devices at a global scale over any compatible medium (currently TCP, WebSockets, Bluetooth Low Energy etc), allowing multi-hop peer-to-peer connectivity between devices even in places where there is no Internet connectivity.

https://github.com/matrix-org/pinecone/

Pinecone supports passing Matrix messages over a BLE mesh network. It would be awesome is Meshtastic could either support pinecone or at least allow running pinecone traffic over Meshtastic. That would then allow decentralised Matrix communication over LoRa

caveman99, 2023-04-24:
From my first glance, i stumbled across this: "The only requirements for a peering today are that it is stream-oriented and reliable;"

Meshtastic is neither, and from the design document this will need way more bandwidth than we can do with meshtastic.

alistair23, 2023-04-28:
No worries. For anyone who stumbles across this, there is at least bridging available, see https://github.com/geoffwhittington/meshtastic-matrix-relay