meshtastic/firmware #2447: [Question] Boot up problem

TRAHOMOTO, 2023-04-25:
My device is **T-Echo** flashed with **2.1.5.23272da** fw. At some moment right after click on "Button 2" (_turn on_) device start to blink with red-blue led  and that is it, no reaction on display, no bluetooth connection just nothing.

After digging a bit, I found these logs on serial port. And i cant find myself any helpful info.

**Could you please help me to fix this problem**

1. I suspect it may happens after I connected device to the power bank
2. Am I understand correctly, the last logs row mean that FS of internal memory were corrupted?
3. Is there any way to wipe T-Echo completely before flashing?

```
Reconnecting to COM8     Connected!
��␀␂@␁INFO  | ??:??:?? 1 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1  channels.proto (90 Bytes)
DEBUG | ??:??:?? 1  db.proto (498 Bytes)
DEBUG | ??:??:?? 1 Using analog input 4 for battery level
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x51
INFO  | ??:??:?? 1 PCF8563 RTC found
INFO  | ??:??:?? 1 1 I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=7, swver=2.1.5.23272daf
DEBUG | ??:??:?? 1 Reset reason: 0x0
DEBUG | ??:??:?? 1 Setting random seed 542106514
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
ERROR | ??:??:?? 1 assert failed C:\Users\Admin\.platformio\packages\framework-arduinoadafruitnrf52\libraries\Adafruit_LittleFS\src\littlefs\lfs.c: 1144
```

caveman99, 2023-04-25:
https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase

TRAHOMOTO, 2023-04-25:
@caveman99 thank you very much. It helps. I missed this article 🤕

closed 2023-04-25 by TRAHOMOTO