meshtastic/firmware #2448: [Bug]: Firmware 2.1.5.23272da on G1 Explorer unable to send packets

alistair23, 2023-04-27:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.1.5.23272da

### Description

When trying to send packets I am getting this error and nothing is being seen by other nodes:

```
DEBUG | 04:03:26 61 [Router] Reliable send failed, returning a nak for fr=0x0,to=0xffffffff,id=0x47cbe01
ERROR | 04:03:26 61 [Router] Alloc an err=5,to=0xef5aa1c8,idFrom=0x47cbe01,id=0x2ed7bd70
```

### Relevant log output

```Shell
//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (479232/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (73 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (149 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (253 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (67 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 0  /static/index-0e816946.js.gz (399863 Bytes)
DEBUG | ??:??:?? 0  /static/index-9602d760.css.gz (17494 Bytes)
DEBUG | ??:??:?? 0  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   490][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   504][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 sh1106 display found
DEBUG | ??:??:?? 0 0x0 subtype probed in 1 tries 
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 Meshtastic hwvendor=17, swver=2.1.5.23272da
DEBUG | ??:??:?? 0 Setting random seed 1977061740
DEBUG | ??:??:?? 0 Total heap: 191392
DEBUG | ??:??:?? 0 Free heap: 156160
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 192, FreeEntries 438, AllEntries 630, NameSpaces 7
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 25
DEBUG | ??:??:?? 0 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 20
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 20
INFO  | ??:??:?? 0 Loading /prefs/module.proto
[... 438 lines cut ...]
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:32 67 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:32 67 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:34 69 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:34 69 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:36 71 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:36 71 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:38 73 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:38 73 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:40 75 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:40 75 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:42 77 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:42 77 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:45 79 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:45 79 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
[HTTPS:I] Request: GET /api/v1/fromradio?all=false (FID=52)
DEBUG | 04:03:46 81 [WebServerThread] webAPI handleAPIv1FromRadio
DEBUG | 04:03:46 81 [WebServerThread] webAPI handleAPIv1FromRadio, len 0
```

alistair23, 2023-04-27:
They are all the same region and LoRa settings, still nothing

caveman99, 2023-04-27:
can you try the 2.1.19 release? 2.1.5 is not exactly current (yes, it is marked stable) and this release was during a bit of a radiolib hiccup with the streaming interface of SX1262. I'm just curious if this will solve stuff.

alistair23, 2023-04-28:
I have tried 2.1.9, same issue

GUVWAF, 2023-04-28:
How close are the devices to each other? A while ago someone couldn't receive with devices in the same room. 

Are you able to get logs of a receiving device when you're transmitting?

alistair23, 2023-04-28:
Ok! So I went back to `firmware-2.0.21.83e6cea` which I think is what ships on the Nano G1 Explorer and I am able to send/receive to my RAK4631 on 2.1.9.

I was having a few reliability issues. It seems like updating to 2.1.9 on the RAK4631 helped. It also seems like the web interface can get kind of stuck which wasn't helping. Still not super reliable though

GUVWAF, 2023-04-28:
And using 2.1.5 or 2.1.9 on the Nano G1 Explorer still doesn't work? 
Mixing firmware 2.0 and 2.1 may indeed be unreliable.
Regarding the Web UI; maybe you can try either the Android or Apple apps or the Python or C# CLI?

alistair23, 2023-04-28:
Here are the logs of a send/receive. Now that I know at least the hardware works (and my setup is correct). I am updating everything to 2.1.9 and moving them to different sides of the house.

Then I'll see if I can narrow down a single failure and report back. Thanks for all of your help so far

[recv.log](https://github.com/meshtastic/firmware/files/11352751/recv.log)
[send.log](https://github.com/meshtastic/firmware/files/11352753/send.log)

alistair23, 2023-04-28:
Thanks for all of your help! With the devices in separate rooms and using the Android application instead of the web client via serial interface communication seems to be working now

The range tester doesn't seem to be running, but that's a different problem

closed 2023-04-28 by alistair23