meshtastic/firmware #2449: [Feature Request]: W5500 Ethernet module

Fertrius, 2023-04-27:
### Platform

NRF52

### Description

Hello, in purpose to have possibility use wired connection instead of WiFi and deliver power to remote unit using various PoE ability, such as passive PoE, PoE splitters... my question:

Is it possible to add support of W5500 Ethernet module to T-Beam or DIY versions based on ESP32/ESP32-S3?
Does SPI bus will allow it with LoRa connected on?

Thank you in advance!

caveman99, 2023-04-27:
duplicate of  #2107 - ESP32 can not do this due to lack of GPIO. S3 is still very new, there's other fish to fry first.

closed 2023-04-27 by caveman99

petrkr, 2024-05-14:
> duplicate of #2107 - ESP32 can not do this due to lack of GPIO. S3 is still very new, there's other fish to fry first.

That is not true, ESP32 have lot of pins to do even 9pin PHY LAN, check proof-of-concept at #2107