meshtastic/firmware #2455: [Bug]: flash installation failed on LORA32 (version: Tiora-v2-1-1.6)

JosefSo, 2023-05-02:
### Category

Serial

### Hardware

T-Lora v2 1.6

### Firmware Version

Tiora-v2-1-1.6

### Description

When I tried to flash this device via browser (Google Chrome) I've got issue (See the screenshot):
[image]

### Relevant log output

```Shell
installation failed
⚠️
Your ESP32-S3 board is not supported.
```

JosefSo, 2023-05-03:
I checked and you are right it is new T3S3. How do I flash it in this case?

megabyte2010, 2023-05-04:
> I checked and you are right it is new T3S3. How do I flash it in this case?

Press boot button, keep it pressed and press and release reset button and release boot button. And you need to flash with cli. The browser flasher gives error

caveman99, 2023-05-23:
This is fixed by now in the web flasher. Not a firmware issue.

closed 2023-05-23 by caveman99