meshtastic/firmware #2460: [Bug]: setting node to REPEATER mode results in endless reboots

mverch67, 2023-05-03:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.1.10

### Description

Setting the node into device.role REPEATER asserts after reboot and reboots again and again...

### Relevant log output

```Shell
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
INFO  | ??:??:?? 1 Loaded saved devicestate version 20
INFO  | ??:??:?? 1 Loading /prefs/config.proto
INFO  | ??:??:?? 1 Loaded saved config version 20
INFO  | ??:??:?? 1 Loading /prefs/module.proto
INFO  | ??:??:?? 1 Loaded saved moduleConfig version 20
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
INFO  | ??:??:?? 1 Loaded saved channelFile version 20
[  1391][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 1 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 1 Number of Device Reboots: 1505
INFO  | ??:??:?? 1 Wanted region 3, using EU_868
DEBUG | ??:??:?? 1 region=3, NODENUM=0x55c607dc, dbsize=8
INFO  | ??:??:?? 1 Saving /prefs/db.proto

assert failed: Router::Router() Router.cpp:60 (!cryptLock)

Backtrace: 0x400845b5:0x3ffd8930 0x40097c0d:0x3ffd8950 0x4009e0e5:0x3ffd8970 0x400eab89:0x3ffd8aa0 0x400e4b01:0x3ffd8ac0 0x400e3d7f:0x3ffd8ae0 0x4013db46:0x3ffd8b50
```

thebentern, 2023-05-03:
Decoded backtrace:
0x400845b5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
0x40097c0d: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
0x4009e0e5: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
0x400eab89: std::__detail::_Node_iterator_base, true>::_M_incr() at /home/runner/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/hashtable_policy.h:300
0x400eab89: std::__detail::_Node_iterator, false, true>::operator++(int) at /home/runner/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/hashtable_policy.h:363
0x400eab89: ReliableRouter::send(_meshtastic_MeshPacket*) at /home/runner/work/firmware/firmware/src/mesh/ReliableRouter.cpp:30
0x400e4b01: Channels::getPrimaryName() at /home/runner/work/firmware/firmware/src/mesh/Channels.cpp:311
0x400e3d7f: setup() at /home/runner/work/firmware/firmware/src/main.cpp:370
0x4013db46: uart_ll_get_txfifo_len at /home/runner/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/hal/esp32/include/hal/uart_ll.h:321
0x4013db46: uart2_write_char at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:499

closed 2023-05-04 by caveman99

daniel-hofer, 2023-05-08:
Hi, tested firmware 2.1.10 with tbeam and heltec v3 in router mode and got attached logs
[heltecv3.txt](https://github.com/meshtastic/firmware/files/11425333/heltecv3.txt)
I think the owner-name was reset to default too.

tavdog, 2023-09-26:
This just happened to me on the Station G1 with beta firmware.

commit 313860c8 (2023-05-04): fix #2460 - we only really need the router object after nodedb init, so lets move it there.