meshtastic/firmware #2466: [Bug]: New Channel on NodeInfo is usually not correct.

garthvh, 2023-05-05:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.1.xx

### Description

The new channel index provided on node info is not proving to be very accurate.

I would think the channel number should be different than the default of 0 is the two nodes share a secondary but not primary channel on the network where the node info's channel > 0

Instead what happens is that usually a node is misidentified as having come from another network when it is local.

On iOS the channel displays on a node if it is anything other than 0

garthvh, 2023-05-05:
Tagging @sbias since I can't find him in the assign search.

caveman99, 2023-05-23:
@sbias did you have a chance to look at this?

garthvh, 2023-05-28:
Going to test this some this week

sbias, 2023-05-28:
i tried to reproduce that with 2.1.6 and python but was not able to reproduce this bug. will try to reproduce it this weekend with a current version again.

sbias, 2023-05-30:
sadly i can not reproduce this. @garthvh can you provide more information how to test this?

garthvh, 2023-06-01:
I just set up a new QR code for our dev mqtt and am seeing nodes with the wrong channel identified

garthvh, 2023-06-01:
[image]

in this example the connected node should not be channel 4

garthvh, 2023-06-04:
I am on MQTT but I am seeing the issue with nodes on my local mesh.

I set up a secondary channel called LongFast and set lora channel 20 I can text back and forth across the networks and eventually everyone gets a node info but I am still seeing some cases where a node on my local network has a node info from the channel id of the longfast network.

sbias, 2023-06-05:
okey, i guess the local node will change its channel each time its sends out its node-info. 

NodeInfoModule::sendOurNodeInfo calls service.sendToMesh(p) calls nodeDB.updateFrom(*p) which updates the NodeDB. But that does not explain wrong channels of remote nodes yet.

garthvh, 2023-09-29:
I am still seeing almost everything over channel 0

garthvh, 2023-10-01:
This remains broken and needs to get fixed, should be relatively easy to get the channel right once the packet is on the device right?  This is causing messages to get delivered to the wrong channels.

sbias, 2023-10-01:
Can you provide a scenario how to reproduce this behavior please? I only have a few nodes and wasn't able to check this. The node should keep track on which channel node-info was heard and store that into the database. a direct message should be assigned to that stored channel while sending.

garthvh, 2023-10-01:
Here is the QR code 
[image]

garthvh, 2023-10-01:
Use that qr and connect to the public mqtt

sbias, 2023-10-05:
One problem - we can not differ between channel unknown and channel 0 . Needs to be fixed. But that does not explain chat wrong node on ch4 in you comment on Jun 1.
I'm on it.

garthvh, 2023-10-07:
Let me know when you need help testing.

sbias, 2023-10-07:
if you like you can try it. should make things better, but it does not explain the channel 4 issue mentioned above. will look over it again after a sleep :)

sbias, 2023-10-08:
ok, i pushed a small fix right now.  i don't like the idea that the mqtt topic has no impact on channel-selection, just the encryption-key - even the channel-name is the topic itself. but changing  that will need some more refactoring. this fix seems to work for now.

garthvh, 2023-10-08:
It seems much better, messages seem to come in on the right channel now.  Most node info's seem right though this should really never happen. 
[image]

garthvh, 2023-10-08:
It does seem to get corrected, but it is frequently wrong when the node info comes in initially with the want config.

sbias, 2023-10-08:
you mean the channel of the node itself is hopping? this will happen everytime the node sends out a nodeInfo packet. And since the node "hears" itself all all channels it knows it will represent the last channel it sent a node-info.
does this cause any problem i don't see right now? of cause, we can just exclude the node itself from updating the database.

andrekir, 2023-10-15:
@garthvh are these NodeInfos comming from MQTT? if so, this could be fixed with #2892

garthvh, 2023-10-15:
I need to test again, most of these nodes are from the public mqtt, but my local nodes have been on the wrong channel.

sbias, 2023-10-16:
Hummm ... #2892 shows an interesting protocol issue with multiple unencrypted channels (or having the same key). Even we have an ServiceEnvelope on mqtt, we dont have that on radio, so channel-assignments are based on bruteforcing all known keys and assign it on success.
@garthvh are all channels you are using have a different encryption-key?

andrekir, 2023-10-17:
the channel hash uses name+PSK, so same PSK and different names are fine. `decryptforHash()` uses [getHash()](https://github.com/meshtastic/firmware/blob/master/src/mesh/Channels.cpp#L296) to look up the [hashes](https://github.com/meshtastic/firmware/blob/master/src/mesh/Channels.h#L122) for each local channel. decoding is attempted only when both match.

the issue comes up later in the code, seems to fail finding/changing the node's channel or persisting it to the nodeDB.

sbias, 2023-10-19:
thanks for correcting me, found it in [Channels::generateHash](https://github.com/meshtastic/firmware/blob/master/src/mesh/Channels.cpp#L31)

sbias, 2023-11-11:
i guess the issue is that the channel index is part of the packet itself which will be wrapped into an encrypted packet. i moved the line to overwrite the index down after decryption to prove this. even it might solve the issue for encrypted channels, it will not work for open-channels.

GUVWAF, 2024-01-13:
One thing I observed that also might explain why the channel is incorrect is when you e.g. make your primary private and move the default to a secondary, other nodes do not receive your periodic NodeInfo anymore. If they already had you in the NodeDB, they will not request a NodeInfo from you when they hear you sending a message and then your channel never gets updated. It will be fixed when the other node deletes you from the NodeDB. 

Another way this can be fixed is by sending your NodeInfo to all the channels once after you change the configuration. Not sure if this is worth it though.

andrekir, 2024-01-13:
just ran a quick test with MQTT LongFast on channel index 6. waited for a few nodes to pop up, then disconnected/connected to see the NodeDB coming from the radio.

[image]

only 1 of 10 nodes had the channel properly set to 6 and... strangely, my local RAK node (that was connected to MQTT) was tagged with channel 6... very weird.

so I don't think it's some edge case we're missing, there's something fundamental not working here.

garthvh, 2024-01-13:
Yeah mqtt seems to be the worst, with just lora nodes it is mostly accurate, though the connected node seems to get confused.

tavdog, 2024-01-13:
I'm seeing this pretty consistently over LoRa with a private key primary channel (0) , admin channel on 1 and public LongFast on 2.  When I get an info packet from a node with their primary channel public longfast they don't seem to get assigned to my local channel 2 as it should.  When I try to direct message them it goes over my primary encrypted channel not my public channel.

tavdog, 2024-01-14:
I managed to reproduce this bug and it involves a restart of the node. The first time a nodeinfo packet comes in the channel is mapped correctly and a private message will send out on the correct channel.  Then after the sending node restarts the channel is gone from the node DB so the private message fails because it sends out on the wrong channel.

GUVWAF, 2024-01-14:
Indeed with MQTT it’s easy to reproduce.

@tavdog Indeed the channel in the nodeDB is not written to flash unless the `user` field is different. I changed updating the channel to within the `updateUser` function of the NodeInfoModule: https://github.com/meshtastic/firmware/pull/3094 

Now for every node the channel is correct (channel 1 in my case) after I disconnect and reconnect: 

https://github.com/meshtastic/firmware/assets/78759985/5dacbce1-1fda-4683-a22b-e138caeb24bb

andrekir, 2024-01-14:
that's it. well done, @GUVWAF !

tavdog, 2024-01-14:
Preliminary testing show this is working.  
[image]

PR #3094 (2024-01-16): Update channel of node in `updateUser` and write to flash if needed
Potentially fixes #2466, see my comment there.
closed 2024-01-16 by thebentern