meshtastic/firmware #2468: [Board]: ESP32S3+RFM95+ST7789+QWERTY keyboard

bobricius, 2023-05-05:
### SOC

ESP32

### Lora IC

RFM95

### Product Link

http://pemi.technology/data/_uploaded/media/picomputer28.png

### Description

Hi, I successfully tested replace RP2040 by BPI-PicoW-S3 which is based on ESP32S3
Display is enhanced to 2.8" IPS, color, 320x240
RFM95 radio 

[https://circuitpython.org/board/bpi_picow_s3/](https://circuitpython.org/board/bpi_picow_s3/)

I tested my software and lora+display+keyboard all working, also I send message to another rp2040 device

**I offer a reward of $300 for a Meshtastic port + I send assembled device HW for testing.**

I provide schematic and more details.

there is something for start [https://github.com/bobricius/Armachat-circuitpython](https://github.com/bobricius/Armachat-circuitpython)

caveman99, 2023-05-06:
I'll take that on, should not be to difficult. I suggest donating the $300 bounty to the Meshtastic org instead :-) If @markbirss wants to have a go, he has first dibs though ...

markbirss, 2023-05-06:
@caveman99 

Keep us posted how support progresses.

bobricius, 2023-05-07:
whoever takes care of it, please let me know the postal address (you can write to me at my private address bobricius@gmail.com) so I can send the device.

Here is universal schematic https://github.com/bobricius/Armachat-circuitpython/blob/main/Hardware/PICOmputer%20-%20Schematic.png

if you have any questions, let me know, almost all my designs use this scheme, the difference is basically only in the layout of the buttons and the display

tested with this board https://wiki.banana-pi.org/BPI-PicoW-S3

**I will send a reward according to the port author's wishes**

caveman99, 2023-08-07:
This is supported now and will be in 2.2.0 firmware release.

closed 2023-08-07 by caveman99