meshtastic/firmware #2474: [Bug]: Packet transmission is delayed indefinitely by receiving on serial

rdavyd, 2023-05-09:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

2.1.11.ed58cda2

### Description

I'm trying to transmit serial data which comes at speed of 2-4 pps, packet size is 200 bytes (this is irrelevant, 20 bytes yield the same result). Radio does not send anything and only receives on serial, until I stop sending packets to it. Only then it tries to transmit.

Serial config:
```C
    moduleConfig.serial.mode = meshtastic_ModuleConfig_SerialConfig_Serial_Mode_SIMPLE;
    moduleConfig.serial.enabled = 1;
    moduleConfig.serial.rxd = 35;
    moduleConfig.serial.txd = 15;
    moduleConfig.serial.timeout = 1000;
    moduleConfig.serial.baud = meshtastic_ModuleConfig_SerialConfig_Serial_Baud_BAUD_115200;
    moduleConfig.serial.echo = 0;
```

In addition to hard coded serial configuration also disabled ACK in `SerialModuleRadio::sendPayload`
<details>
  <summary>SerialModuleRadio::sendPayload</summary>

```C
void SerialModuleRadio::sendPayload(NodeNum dest, bool wantReplies)
{
    meshtastic_Channel *ch = (boundChannel != NULL) ? &channels.getByName(boundChannel) : NULL;
    meshtastic_MeshPacket *p = allocReply();
    p->to = dest;
    if (ch != NULL) {
        p->channel = ch->index;
    }
    p->decoded.want_response = wantReplies;

    //ACK disabled
    p->want_ack = 0;

    p->decoded.payload.size = serialPayloadSize; // You must specify how many bytes are in the reply
    memcpy(p->decoded.payload.bytes, serialBytes, p->decoded.payload.size);

    service.sendToMesh(p,RX_SRC_LOCAL,false);
}
```
</details>

### Relevant log output

```Shell
<details>
  <summary>Logs</summary>

--- Terminal on COM11 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_c3_exception_decoder, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
DEBUG | ??:??:?? 162 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
DEBUG | ??:??:?? 182 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
DEBUG | ??:??:?? 190 setting notification 1
DEBUG | ??:??:?? 190 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=101, time 135 ms
ERROR | ??:??:?? 190 [RadioIf] ignoring received packet due to error=-7
DEBUG | ??:??:?? 190 [RadioIf] AirTime - Packet received (noise?) : 135ms
DEBUG | ??:??:?? 192 [SerialModule] Update DB node 0xe56a710, rx_time=0, channel=0
[... 649 lines cut ...]
DEBUG | ??:??:?? 210 [RadioIf] delaying notification 0
DEBUG | ??:??:?? 210 [RadioIf] Starting low level send (id=0x7d6a9bdc fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 210 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=221, time 275 ms
DEBUG | ??:??:?? 210 [RadioIf] AirTime - Packet transmitted : 275ms
DEBUG | ??:??:?? 211 setting notification 2
DEBUG | ??:??:?? 211 [RadioIf] Completed sending (id=0x7d6a9bdc fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 211 [RadioIf] xmit timer 42
DEBUG | ??:??:?? 211 [RadioIf] setting notification 3
DEBUG | ??:??:?? 211 [RadioIf] delaying notification 42
DEBUG | ??:??:?? 211 [RadioIf] Starting low level send (id=0x7d6a9bde fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 211 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=221, time 275 ms
DEBUG | ??:??:?? 211 [RadioIf] AirTime - Packet transmitted : 275ms
DEBUG | ??:??:?? 211 setting notification 2
DEBUG | ??:??:?? 211 [RadioIf] Completed sending (id=0x7d6a9bde fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 211 [RadioIf] xmit timer 42
DEBUG | ??:??:?? 211 [RadioIf] setting notification 3
DEBUG | ??:??:?? 211 [RadioIf] delaying notification 42
DEBUG | ??:??:?? 211 [RadioIf] Starting low level send (id=0x7d6a9be0 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 211 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=221, time 275 ms
DEBUG | ??:??:?? 211 [RadioIf] AirTime - Packet transmitted : 275ms
DEBUG | ??:??:?? 211 setting notification 2
DEBUG | ??:??:?? 211 [RadioIf] Completed sending (id=0x7d6a9be0 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)

</details>
```

rdavyd, 2023-05-10:
Setting serial timeout to 1 somewhat helped. But still messages in transmit queue are accumulating faster than they are transmitted.

<details>
  <summary>Logs</summary>

```bash
DEBUG | ??:??:?? 490 [RadioIf] Completed sending (id=0x010ad208 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 490 [RadioIf] xmit timer 84
DEBUG | ??:??:?? 490 [RadioIf] setting notification 3
DEBUG | ??:??:?? 490 [RadioIf] delaying notification 84
DEBUG | ??:??:?? 491 [SerialModule] Update DB node 0xe56a710, rx_time=0, channel=0
DEBUG | ??:??:?? 491 [SerialModule] localSend to channel 0
DEBUG | ??:??:?? 491 [SerialModule] Add packet record (id=0x010ad211 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x0 Portnum=64)
WARN  | ??:??:?? 491 [SerialModule] User provided a too short AES256 key - padding
DEBUG | ??:??:?? 491 [SerialModule] Using AES256 key!
DEBUG | ??:??:?? 491 [SerialModule] ESP32 crypt fr=e56a710, num=10ad211, numBytes=24!
DEBUG | ??:??:?? 491 [SerialModule] enqueuing for send (id=0x010ad211 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted)
DEBUG | ??:??:?? 491 [SerialModule] txGood=17,rxGood=0,rxBad=0
INFO  | ??:??:?? 491 [SerialModule] New TX queue size 9
DEBUG | ??:??:?? 491 [SerialModule] xmit timer 126
DEBUG | ??:??:?? 491 [SerialModule] dropping notification 3
DEBUG | ??:??:?? 491 [RadioIf] Starting low level send (id=0x010ad209 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 491 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=40, time 63 ms
DEBUG | ??:??:?? 491 [RadioIf] AirTime - Packet transmitted : 63ms
DEBUG | ??:??:?? 491 setting notification 2
DEBUG | ??:??:?? 491 [RadioIf] Completed sending (id=0x010ad209 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 491 [RadioIf] xmit timer 84
DEBUG | ??:??:?? 491 [RadioIf] setting notification 3
DEBUG | ??:??:?? 491 [RadioIf] delaying notification 84
DEBUG | ??:??:?? 491 [SerialModule] Update DB node 0xe56a710, rx_time=0, channel=0
DEBUG | ??:??:?? 491 [SerialModule] localSend to channel 0
DEBUG | ??:??:?? 491 [SerialModule] Add packet record (id=0x010ad212 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x0 Portnum=64)
WARN  | ??:??:?? 491 [SerialModule] User provided a too short AES256 key - padding
DEBUG | ??:??:?? 491 [SerialModule] Using AES256 key!
DEBUG | ??:??:?? 491 [SerialModule] ESP32 crypt fr=e56a710, num=10ad212, numBytes=24!
DEBUG | ??:??:?? 491 [SerialModule] enqueuing for send (id=0x010ad212 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted)
DEBUG | ??:??:?? 491 [SerialModule] txGood=18,rxGood=0,rxBad=0
INFO  | ??:??:?? 491 [SerialModule] New TX queue size 9
DEBUG | ??:??:?? 491 [SerialModule] xmit timer 84
DEBUG | ??:??:?? 491 [SerialModule] dropping notification 3
DEBUG | ??:??:?? 491 [SerialModule] Update DB node 0xe56a710, rx_time=0, channel=0
DEBUG | ??:??:?? 491 [SerialModule] localSend to channel 0
DEBUG | ??:??:?? 491 [SerialModule] Add packet record (id=0x010ad213 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x0 Portnum=64)
WARN  | ??:??:?? 491 [SerialModule] User provided a too short AES256 key - padding
DEBUG | ??:??:?? 491 [SerialModule] Using AES256 key!
DEBUG | ??:??:?? 491 [SerialModule] ESP32 crypt fr=e56a710, num=10ad213, numBytes=24!
DEBUG | ??:??:?? 491 [SerialModule] enqueuing for send (id=0x010ad213 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted)
DEBUG | ??:??:?? 491 [SerialModule] txGood=18,rxGood=0,rxBad=0
INFO  | ??:??:?? 491 [SerialModule] New TX queue size 10
DEBUG | ??:??:?? 491 [SerialModule] xmit timer 84
DEBUG | ??:??:?? 491 [SerialModule] dropping notification 3
DEBUG | ??:??:?? 491 [RadioIf] Starting low level send (id=0x010ad20a fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 491 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=40, time 63 ms
DEBUG | ??:??:?? 491 [RadioIf] AirTime - Packet transmitted : 63ms
DEBUG | ??:??:?? 491 setting notification 2
DEBUG | ??:??:?? 491 [RadioIf] Completed sending (id=0x010ad20a fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 491 [RadioIf] xmit timer 42
DEBUG | ??:??:?? 491 [RadioIf] setting notification 3
DEBUG | ??:??:?? 491 [RadioIf] delaying notification 42
DEBUG | ??:??:?? 491 [SerialModule] Update DB node 0xe56a710, rx_time=0, channel=0
DEBUG | ??:??:?? 491 [SerialModule] localSend to channel 0
DEBUG | ??:??:?? 491 [SerialModule] Add packet record (id=0x010ad214 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x0 Portnum=64)
WARN  | ??:??:?? 491 [SerialModule] User provided a too short AES256 key - padding
DEBUG | ??:??:?? 491 [SerialModule] Using AES256 key!
DEBUG | ??:??:?? 491 [SerialModule] ESP32 crypt fr=e56a710, num=10ad214, numBytes=24!
DEBUG | ??:??:?? 491 [SerialModule] enqueuing for send (id=0x010ad214 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted)
DEBUG | ??:??:?? 491 [SerialModule] txGood=19,rxGood=0,rxBad=0
INFO  | ??:??:?? 491 [SerialModule] New TX queue size 10
DEBUG | ??:??:?? 491 [SerialModule] xmit timer 84
DEBUG | ??:??:?? 491 [SerialModule] dropping notification 3
[... 918 lines cut ...]
DEBUG | ??:??:?? 501 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=40, time 63 ms
DEBUG | ??:??:?? 501 [RadioIf] AirTime - Packet transmitted : 63ms
DEBUG | ??:??:?? 501 setting notification 2
DEBUG | ??:??:?? 501 [RadioIf] Completed sending (id=0x010ad239 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
[... cut ...]
DEBUG | ??:??:?? 502 [RadioIf] (bw=250, sf=7, cr=4/8) packet symLen=0 ms, payloadSize=40, time 63 ms
DEBUG | ??:??:?? 502 [RadioIf] AirTime - Packet transmitted : 63ms
DEBUG | ??:??:?? 502 setting notification 2
DEBUG | ??:??:?? 502 [RadioIf] Completed sending (id=0x010ad243 fr=0x10 to=0xff, WantAck=0, HopLim=1 Ch=0x5d encrypted priority=64)
DEBUG | ??:??:?? 502 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
DEBUG | ??:??:?? 522 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0
```

</details>

garthvh, 2023-05-17:
Is PPS packets per second?

garthvh, 2023-05-17:
If so you probably need to try something more like 8 seconds between each transmission, especially if they are large packets.

rdavyd, 2023-05-18:
@garthvh I figured, that default LORA channel bandwidth 128kHz is not enough to pass all data. However it didn't even try to send and was busy on receiving serial. Nevertheless I will try 250 and 500 kHz BWs.

closed 2023-10-10 by thebentern