meshtastic/firmware #2479: [Feature Request]: Support NeighborInfo packet in firmware

ajmcquilkin, 2023-05-10:
### Platform

NRF52, ESP32

### Description

This is intended as a tracking ticket for implementing the functionality discussed in this [protobuf PR](https://github.com/meshtastic/protobufs/pull/341).

A quick summary of the discussion:
- The `NeighborInfo` packets will need to be opt-in and respect the user's current location settings
- `NeighborInfo` packets will need to have a configurable transmission interval
- `NeighborInfo` packets will **not** be stored within the NodeDB
- There is discussion around including the most recent forwarder of a given packet within the packet header, but this will require breaking changes and is not a high priority
- Clients will need to individually set the timeout duration for incoming `NeighborInfo` packets, since the `rxTime` and `txTime` fields were removed from the original protobuf implementation
- Nodes will need to only transmit a maximum of `n` most relevant edges within any given `NeighborInfo` packet, which should likely be determined by the quality of the signal
- A "smart broadcast" feature wouldn't work without the `rxTime` and `txTime` fields, since clients wouldn't know whether an edge hasn't changed or whether the edge has dropped

I don't have the bandwidth to work these changes at this moment, but putting this up in case someone's interested in taking them on.

caveman99, 2023-05-25:
"High Priority" since this is supporting the desktop management client which is a game changer for (not only) professional users.

GUVWAF, 2023-05-28:
There is already an implementation of this [here](https://github.com/uhuruhashimoto/firmware/pull/7). Is it just a matter of merging this or is there more to it?

ajmcquilkin, 2023-05-28:
I had worked with Uhuru on that PR a while back, but those changes don't fit within the new specifications. For example, that PR uses the NodeDB to store neighbors, and also uses the header to track packets. I believe it also uses an older forked version of the `NeighborInfo` message that relies on the additional timestamps that were removed in [this PR](https://github.com/meshtastic/protobufs/pull/341). I think that PR can certainly serve as a reference, but it shouldn't be considered at all production-ready at this point.

caveman99, 2023-05-31:
see https://github.com/meshtastic/firmware/pull/2535

ajmcquilkin, 2023-08-08:
Is this issue good to close?

garthvh, 2023-08-11:
> Is this issue good to close?

If you are happy with this as a baseline yes.

closed 2023-08-12 by thebentern