meshtastic/firmware #2481: [Feature Request]: Managed Radio Device Role

garthvh, 2023-05-10:
### Platform

NRF52, ESP32

### Description

It would be nice if we had a role for managed radio. This would be easy to implement on clients as we can just hide the settings tabs if the user is in the managed role.

* [ ] New Device Role Value
* [ ] Code to prevent local config changes
* [ ] All device configuration would take place via the admin channel

tropho23, 2023-05-10:
Excellent idea!

ajmcquilkin, 2023-05-11:
Seems like a great idea! Would a managed radio be a radio that is intended completely for remote administration, or would this be more of a node that isn't intended to be configured after it's initially set up?

garthvh, 2023-05-11:
The two main requests have been for a simplified app without settings for end users, and a way for the admin channel to be used without the end user nodes being able to administer all the nodes on the network.

ajmcquilkin, 2023-05-11:
Okay, so this would essentially be a UI block, or would the device reject configuration from anywhere but the admin channel?

caveman99, 2023-05-11:
i'd make it a setting rather than a role. cause role alters device behaviour. What if i want a managed tracker? - It would be a no-return setting like the serial console switchoff. :-)

imBinky, 2023-05-11:
brane

garthvh, 2023-05-11:
> i'd make it a setting rather than a role. cause role alters device behaviour. What if i want a managed tracker? - It would be a no-return setting like the serial console switchoff. :-)

This is a good point.

PR #2487 (2023-05-12): Adding device.is_managed protobuf
Fixes #2481