meshtastic/firmware #2482: [Bug]: Unable to flash M5Stack Core2

loganmarchione, 2023-05-11:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

N/A

### Description

I'm new to Meshtastic, so bear with me here.

I have a [M5Stack Core2](https://shop.m5stack.com/collections/m5-controllers/products/m5stack-core2-esp32-iot-development-kit). I saw a variant [here](https://github.com/meshtastic/firmware/tree/master/variants/m5stack_core) and also found [this](https://github.com/meshtastic/firmware/blob/666a1f3401b2da2ff9facf6c0227cd2622777bea/src/mesh/generated/meshtastic/mesh.pb.h#L86), so I figured I could flash it. I'm running Arch Linux and it shows up under `/dev/ttyACM0`.

```
> esptool.py --port /dev/ttyACM0 chip_id
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:25:ef:9c
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 30:c6:f7:25:ef:9c
Hard resetting via RTS pin...
```

I cloned this repo, built the firmware according to [these](https://meshtastic.org/docs/development/firmware/build) instructions, but had to change the [port](https://github.com/meshtastic/firmware/blob/666a1f3401b2da2ff9facf6c0227cd2622777bea/variants/m5stack_core/platformio.ini#L5-L6) from `COM8` to `/dev/ttyACM0`. I then tried to flash as per the instructions.  Below is the output.

```
 *  Executing task: platformio run --target upload --environment m5stack-core 

Processing m5stack-core (board: m5stack-core-esp32; platform: platformio/espressif32@^6.1.0; framework: arduino)
----------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-core-esp32.html
PLATFORM: Espressif 32 (6.2.0) > M5Stack Core ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20008.0 (2.0.8) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[nanopb] Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in /home/logan/.platformio/penv/lib/python3.11/site-packages (4.23.0)
[nanopb] No generation needed.
Found 71 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0+sha.b38094e
|-- OneButton @ 2.0.3
|-- arduino-fsm @ 2.2.0+sha.7db3702
|-- TinyGPSPlus @ 1.0.3+sha.127ad67
|-- Thread @ 0.0.0+sha.72921ac
[... 290 lines cut ...]
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x001f3fff...
Compressed 2046000 bytes to 1255303...
Wrote 2046000 bytes (1255303 compressed) at 0x00000000 in 110.6 seconds (effective 147.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION]
                           [--no-progress] [--verify] [--encrypt] [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'bleota.bin'

```

In the end, I had to erase it and was able to flash a demo project again. Am I missing something simple? Again, this my first time using Meshtastic.

```
> esptool.py --port /dev/ttyACM0 erase_flash 
```

mverch67, 2023-05-11:
The M5Stack Core2 has different pin definitions than the old version. So you cannot use the old firmware for the new device. You would need to adapt the pin definitions in the variant.h file according to the Core2 definitions (e.g. MISO is 38):
[image]

Also, when you use device_install.sh you have to specify the firmware.factory.bin file.

loganmarchione, 2023-05-11:
Ahh ok my bad. Thanks for the reply! I have Lilygo device on the way, so I'll wait for that!

closed 2023-05-11 by loganmarchione

caveman99, 2023-05-11:
I wrote the initial m5stack firmware, and at that ime i thought the core2 had the same pinout than the predecessors. i have a m5stack basic and a 9dop version here, no core2. it should not be too difficult to port meshtastic to the core2, since most of the components are similar and the TFT driver is configured via defines.