meshtastic/firmware #2483: [Bug]: Unable to reconnect to mqtt with tls after wireless beacon timeout

zarya, 2023-05-11:
### Category

WiFi

### Hardware

Not Applicable

### Firmware Version

v1-2.1.11.5ec624d

### Description

Node is unable to reconnect to MQTT (TLS) after wireless reconnection.

### Relevant log output

```Shell
INFO  | 05:35:13 898 [mqtt] Received MQTT topic msh/2/c/<removed>/!<removed>, len=76
[910880][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[910881][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[910884][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
[910894][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():382]: (-78) UNKNOWN ERROR CODE (004E)
DEBUG | 05:35:25 910 WiFi-Event 5: Disconnected from WiFi access point
DEBUG | 05:35:25 910 [Router] Found existing packet record for fr=0x6d00f6f0,to=0xffffffff,id=<removed>
DEBUG | 05:35:25 910 [Router] Found existing packet record for fr=0x6d00f6f0,to=0xffffffff,id=<removed>
DEBUG | 05:35:25 910 [Router] Add packet record (id=<removed> fr=0xf0 to=0xff, WantAck=0, HopLim=6 Ch=0x0 Portnum=67 rxtime=1683783312 priority=1)
DEBUG | 05:35:25 910 [Router] Ignoring incoming msg, because we've already seen it (id=<removed> fr=0xf0 to=0xff, WantAck=0, HopLim=6 Ch=0x0 Portnum=67 rxtime=1683783312 priority=1)
DEBUG | 05:35:25 910 [Router] Incoming message was filtered <removed>
DEBUG | 05:35:25 910 [RadioIf] Completed sending (id=<removed> fr=0xf0 to=0xff, WantAck=0, HopLim=6 Ch=0xee encrypted rxtime=1683783312 priority=1)
INFO  | 05:35:26 911 [WifiConnect] Reconnecting to WiFi access point issues
[917780][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 05:35:32 917 WiFi-Event 4: Connected to access point
[918424][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[918425][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: <removed>, MASK: 255.255.255.0, GW: <removed>
DEBUG | 05:35:32 918 WiFi-Event 7: Obtained IP address: Using TLS-encrypted session
INFO  | 05:35:32 918 Connecting to MQTT server <removed>, port: 8883, username: <removed>, password: <removed>
[918492][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[918493][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
[918500][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -32512
ERROR | 05:35:32 918 Failed to contact MQTT server (1/5)...
```
```

closed 2023-08-14 by garthvh

garthvh, 2024-01-29:
Check your mqtt config. Are you running a private server with a cert for tls?

zarya, 2024-01-29:
Yes i am, server used is EMQX the cert is self signed first connect after reboot it works flawless but after it loses wifi it stops working and trows the mem allocation failed msg.

tavancini, 2024-02-06:
Same here:
T-Beam v1.1 SX1268 433/470Mhz

Trying to connect to a public MQTT server in Brazil:

```
INFO  | 16:44:51 14340 [mqtt] Using TLS-encrypted session
INFO  | 16:44:51 14340 [mqtt] Attempting to connect directly to MQTT server platform.meshbrasil.com, port: 1883, username: meshdev, password: large4cats
[14340499][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[14340500][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
```

oherrala, 2024-04-20:
This might be related to #3496.