meshtastic/firmware #2496: [Feature Request]: Display all received position values in log

prokrypt, 2023-05-16:
### Platform

ESP32

### Description

It would be nice if the serial log outputs all received position values such as PDOP and altitude, whether locally from GPS or from another node.

```
INFO  | 04:09:58 106809 [Router] Received position from=0xb2479080, id=0x30afdcef, portnum=3, payloadlen=26
INFO  | 04:09:58 106809 [Router] POSITION node=b2479080 l=26 LAT LON MSL PDOP SIV TIME 
INFO  | 04:09:58 106809 [Router] updatePosition REMOTE node=0xb2479080 time=1684123799, latI=37xxxxxxx, lonI=-122xxxxxxx
DEBUG | 04:09:58 106809 [Router] Node status update: 3 online, 3 total
```

garthvh, 2023-05-16:
Other than altitude all of the position flags are opt in so they are not sent and do not use any airtime unless they are explicitly enabled.

prokrypt, 2023-05-17:
> Other than altitude all of the position flags are opt in so they are not sent and do not use any airtime unless they are explicitly enabled.

Indeed. It would be nice to at least display the values that have been received or opted in...

tavdog, 2023-09-28:
Alright my first contribution!  Let's close this thang.

closed 2023-09-28 by thebentern