meshtastic/firmware #2503: [Bug]: RAK 4631: Does not recover from dead battery

GaryR-12, 2023-05-21:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.1.9.d43ddc9

### Description

I'm using the RAK WisBlock system with RAK solar panel and a 103450 battery. Before deploying as a router I tested running the battery down until the unit turned off, then  placed it under a lamp and verified that it restarted and began charging. 
After a few days of deployment the battery died and even after 2 days of full sun the unit did not respond. I recovered the unit and again placed it under a lamp. The red LED lit up but the unit would not start. I tried pressing the button but still no startup. I had to connect to USB power to get the unit to boot.

thebentern, 2023-05-21:
Sadly, this is not something we can control with the firmware. As you noted with plugging in the USB, only hardware intervention can wake the unit when it goes into CPU deep sleep. An external MCU can do this, but again that's a hardware solution. 
If you're using a GPS module with that device, I recommend replacing that with a much lower power consumption real time clock to retain accurate time for the device. GPS is too power hungry for compact solar setups.

closed 2023-05-21 by thebentern

andrekir, 2023-05-21:
also relevant https://github.com/meshtastic/firmware/pull/2040

until a reliable batt voltage reading method is found there is no simple fix

GaryR-12, 2023-05-21:
> Sadly, this is not something we can control with the firmware. As you noted with plugging in the USB, only hardware intervention can wake the unit when it goes into CPU deep sleep. An external MCU [...]

Thanks for the information! I am wondering if it cant wake from deep sleep then why is it being put into deep sleep? Why is it being sold as a solar solution? Why was it able to recover several times on it's own before I deployed it?
I am not running any peripherals on the RAK board. Do you have any recommendations for a setup that will work for a solar powered router?