meshtastic/firmware #2504: [Bug]:

TheRealCryss, 2023-05-21:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.1.12

### Description

It isn't obvious _where_ exactly to click in order to save the settings. There should be a visual hint for that when the mouse is not over the area and when it hovers over it as well.

This report is for the current Meshtastic app from the Appstore on MacOS. I haven't tested it on other OSs.

[image]

prokrypt, 2023-05-21:
This belongs @ https://github.com/meshtastic/Meshtastic-Apple/issues

prokrypt, 2023-05-21:
I believe this belongs @ https://github.com/meshtastic/Meshtastic-Apple/issues

TheRealCryss, 2023-05-21:
I'm sorry. Should I delete it or can it be moved?

closed 2023-05-21 by TheRealCryss

garthvh, 2023-05-21:
It is a catalyst issue, hopefully they address it soon, no need to open a bug