meshtastic/firmware #2519: [Bug]: T-Beam Supreme - Display stays black

Tom9876, 2023-05-26:
### Category

Display not working - stay black

### Hardware

LilyGO T-Beam Supreme U-blox MAX-M10S

### Firmware Version

2.1.11.ed58cda2

### Description

The  OLED Display (SH1106 - I2C connected to SDA:1017 / SCL:1018) stays black
Using SoftRF the display works fine
I re-flashed several times (also with the "core-board" disconnected from the "body", using CLI ((“./device-install.sh -f firmware-tbeam-s3-core-2.1.11.5ec624d.bin”) and the web-interface.
Watching the log I noticed **a 2 minute pause after "Scanning for i2c devices on port 1"**. When the "core-board" is disconnected from the "body" there is no pause.
Also setting the display manually to SH1106 does not cure the problem.

### Relevant log output

```
Scanning for i2c devices...
DEBUG | ??:??:?? 2 Scanning for i2c devices on port 2
DEBUG | ??:??:?? 3 I2C device found at address 0x34
INFO  | ??:??:?? 3 axp192/axp2101 PMU found
DEBUG | ??:??:?? 3 I2C device found at address 0x51
INFO  | ??:??:?? 3 PCF8563 RTC found
DEBUG | ??:??:?? 4 Scanning for i2c devices on port 1
INFO  | ??:??:?? 130 2 I2C devices found

```

garthvh, 2023-05-26:
@lewisxhe We are receiving some reports of screen issues with the new Tbeam Supreme

geeksville, 2023-05-26:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-lilygo-t-beam-supreme-product/7474/27

lewisxhe, 2023-05-27:
A fix has been submitted https://github.com/meshtastic/firmware/pull/2520

caveman99, 2023-05-27:
merged

closed 2023-05-27 by caveman99