meshtastic/firmware #2524: [Bug]: T-Beam Supreme - No environmental telemetry data (BMP-280)

Tom9876, 2023-05-27:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.1.14.fbcd6743

### Description

After solving the display issue (see #2519), which was - to my understanding - due to missing power up of the display, all I2C sensors of the T-Beam Supreme were properly detected (thanks for the quick support). For unknown reasons the connection to the BMP-280 failed: "Could not connect to detected BMP280 sensor" (see log). Thus no environmental data available. 
When using SoftRF the sensor works fine.
One more remark: Even though a Micro-SD-Card is inserted (FAT formatted) it is not detected, log: "No SD_MMC card detected".

### Relevant log output

```Shell
INFO  | ??:??:?? 2 Scanning for i2c devices...
DEBUG | ??:??:?? 2 Scanning for i2c devices on port 2
DEBUG | ??:??:?? 3 I2C device found at address 0x34
INFO  | ??:??:?? 3 axp192/axp2101 PMU found
DEBUG | ??:??:?? 3 I2C device found at address 0x51
INFO  | ??:??:?? 3 PCF8563 RTC found
DEBUG | ??:??:?? 4 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 4 I2C device found at address 0x1c
INFO  | ??:??:?? 4 QMC6310 Highrate 3-Axis magnetic sensor found
DEBUG | ??:??:?? 5 I2C device found at address 0x3c
INFO  | ??:??:?? 5 sh1106 display found
DEBUG | ??:??:?? 5 0x0 subtype probed in 1 tries 
DEBUG | ??:??:?? 5 I2C device found at address 0x77
DEBUG | ??:??:?? 5 Wire.available() = 0
INFO  | ??:??:?? 5 BMP-280 sensor found at address 0x77
INFO  | ??:??:?? 6 5 I2C devices found
DEBUG | ??:??:?? 6 acc_info = 0
DEBUG | ??:??:?? 6 found i2c sensor meshtastic_TelemetrySensorType_BMP280
DEBUG | ??:??:?? 6 found i2c sensor meshtastic_TelemetrySensorType_QMC6310
DEBUG | ??:??:?? 6 No SD_MMC card detected
INFO  | ??:??:?? 6 Meshtastic hwvendor=12, swver=2.1.14.fbcd6743
…
….
INFO  | 18:44:54 18 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
INFO  | 18:44:54 18 [EnvironmentTelemetryModule] Init sensor: BMP280
WARN  | 18:44:54 18 [EnvironmentTelemetryModule] Could not connect to detected BMP280 sensor.
 Removing from nodeTelemetrySensorsMap.
```

geeksville, 2023-05-27:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/new-lilygo-t-beam-supreme-product/7474/32

garthvh, 2023-05-27:
Meshtastic has no sd card support

caveman99, 2023-05-28:
Looks like the sensor is assumed on the wrong I2C Bus...

lewisxhe, 2023-05-29:
Fixed by https://github.com/meshtastic/firmware/pull/2529

Tom9876, 2023-05-29:
The fix works for me. Thanks
(according to the LilyGO product description it is a BME280 sensor and not a BMP280)

caveman99, 2023-05-30:
will be fixed in next firmware release. Thanks @lewisxhe

closed 2023-05-30 by caveman99